povik/sv-elab #3: Parsing standard cells

dpetrisko, 2024-06-13:
Hello! I am attempting to use yosys-slang to parse a design with explictly instantiated standard cells: 

```
module bsg_clkbuf #(parameter width_p=1)
   (input    [width_p-1:0] i
    , output [width_p-1:0] o
    );

    for (genvar j = 0; j < width_p; j++)
      sky130_fd_sc_hd__clkbuf_1 b (.X(o[j]), .A(i[j]));

endmodule
```

yosys.tcl
```
yosys -import
plugin -i slang
yosys -import

read_slang --ignore-unknown-modules --timescale=1ps/1ps bsg_clkbuf.sv
```

yosys.tcl
```

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.41 (git sha1 c1ad37779, g++ 11.2.1 -fPIC -Os)
...
1. Executing SLANG frontend.
Top level design units:
    bsg_clkbuf

Build succeeded: 0 errors, 0 warnings
{
  "name": "b",
  "kind": "UninstantiatedDef",
  "addr": 48639872,
  "definitionName": "sky130_fd_sc_hd__clkbuf_1",
  "parameters": [
  ],
  "ports": [
    {
      "name": "X",
      "expr": {
        "kind": "ElementSelect",
        "type": "logic",
        "value": {
          "kind": "NamedValue",
          "type": "logic[0:0]",
          "symbol": "48628160 o"
        },
        "selector": {
          "kind": "NamedValue",
          "type": "integer",
          "symbol": "48630024 j",
          "constant": "0"
        }
      }
    },
    {
      "name": "A",
      "expr": {
        "kind": "ElementSelect",
        "type": "logic",
        "value": {
          "kind": "NamedValue",
          "type": "logic[0:0]",
          "symbol": "48627664 i"
        },
        "selector": {
          "kind": "NamedValue",
          "type": "integer",
          "symbol": "48630024 j",
          "constant": "0"
        }
      }
    }
  ]
}
Source line bsg_clkbuf.sv:8.33:       sky130_fd_sc_hd__clkbuf_1 b (.X(o[j]), .A(i[j]));
ERROR: Feature unimplemented at slang_frontend.cc:1582, see AST and code line dump above
```

The --ignore-unknown-modules is needed to prevent erroring on a missing definition.  If I provide a dummy definition for sky130_fd_sc_hd__clkbuf_1, then the error goes away and it's successfully parsed. For other methods, yosys itself provides the blackboxing based on lib files, but this happens after read_verilog e.g.. What would be the proper method of providing this definition to yosys-slang?

Thanks!

povik, 2024-06-13:
> What would be the proper method of providing this definition to yosys-slang?

We either need a change to Slang which would allow for the provision of external definitions (it would be an API for yosys-slang to use), or we can rely on the `--ignore-unknown-modules` flag and change yosys-slang to convert the `UninstantiatedDef` nodes into netlist cells for Yosys to interpret.

I am inclined toward the latter approach.

povik, 2024-06-13:
> change yosys-slang to convert the UninstantiatedDef nodes into netlist cells for Yosys to interpret.

I have made the change in 28689f9, but yosys-slang still struggles on your example if the `width_p` parameter is more than 1. It fails with

    ERROR: Assert `count_id(wire->name) == 0' failed in kernel/rtlil.cc:2134.

I need to look into that later.

dpetrisko, 2024-06-13:
Thank you for the quick response!

povik, 2024-06-15:
> yosys-slang still struggles on your example if the `width_p` parameter is more than 1. It fails with
> 
> ```
> ERROR: Assert `count_id(wire->name) == 0' failed in kernel/rtlil.cc:2134.
> ```

Should be fixed now (31c464d407edc92de5d977ad34c6a5dfc232db3c and 2d0e7f19d1cc6fd7271e8958811a3cad7518a37f)

closed 2024-06-15 by povik