povik/sv-elab #5: Issue with multiple standard cells with same name

dpetrisko, 2024-06-21:
Hello! There appears to be an issue when having multiple standard cells with the same name, which happens often with wrapper functions.

```
module clkbuf_wrap (input i, output o);

  sky130_fd_sc_hd__clkbuf_1 b (.A(i), .X(o));

endmodule

module test_module (input clk);

  wire test0_clk_buf_lo;
  clkbuf_wrap b0 (.i(clk), .o(test0_clk_buf_lo));
  wire test1_clk_buf_lo;
  clkbuf_wrap b1 (.i(clk), .o(test1_clk_buf_lo));

endmodule
```

```
1. Executing SLANG frontend.
Top level design units:
    tt_um_dpetrisko_ttdll

Build succeeded: 0 errors, 0 warnings
ERROR: Assert `count_id(cell->name) == 0' failed in kernel/rtlil.cc:2152.
```

This works on the other hand:
```
module test_module (input clk);

  wire test0_clk_buf_lo;
  sky130_fd_sc_hd__clkbuf_1 b0 (.i(clk), .o(test0_clk_buf_lo));
  wire test1_clk_buf_lo;
  sky130_fd_sc_hd__clkbuf_1 b1 (.i(clk), .o(test1_clk_buf_lo));

endmodule
```

povik, 2024-06-22:
Thanks for the report! I think with cb80634 this should now be fixed.

closed 2024-06-22 by povik

dpetrisko, 2024-06-24:
Just wanted to share that this solved my issue and I'm able to parse my full design! Thanks for the quick fixes :)