povik/sv-elab #18: Assert failure on SCR1

rubenbuchatskiy, 2024-08-02:
Hello!

When running `yosys-slang` on the `scr1` project, an error is displayed (the assert is triggered).

Steps to reproduce:
```sh
git clone https://github.com/syntacore/scr1.git
cd scr1/src
```

run yosys with slang frontend (yosys-slang version: [67e920e](https://github.com/povik/yosys-slang/commit/67e920ecdc79712af8bf1587a25b1ccd8aca4f23)):
```sh
yosys -m ~/yosys-slang/build/slang.so -p "read_slang  --compat vcs -top scr1_top_ahb -U SCR1_TRGT_SIMULATION -timescale=1ns/1ps -Wno-invalid-source-encoding +incdir+includes +incdir+tb +define+SCR1_TRGT_SIMULATION +define+SCR1_TRACE_LOG_DIS +define+SCR1_CFG_RV32IMC_MAX -f core.files -f ahb_top.files"
```
```sh
ERROR: Assert `!frames.back().locals.count(symbol)' failed in slang_frontend.cc:1001
```

povik, 2024-08-06:
The cause for the failed assertion is addressed in 4c0d6466, I have also implemented the pre-increment operator in 5ee7dbc. With that, the parsing of `scr1` stops on an unimplemented return statement at `core/pipeline/scr1_ipic.sv:96`. I will open an issue to track that support.

I see your affiliation with the Russian academy of sciences. What do you intent on using this tooling for?

povik, 2024-08-06:
Return support: #21

rubenbuchatskiy, 2024-08-06:
> The cause for the failed assertion is addressed in [4c0d646](https://github.com/povik/yosys-slang/commit/4c0d64668dfaecb68afa151e034a679a83047247), I have also implemented the pre-increment operator in [5ee7dbc](https://github.com/povik/yosys-slang/commit/5ee7dbcf4f6a15ff6d5e2d964a5bd73f082588b1). With that, the parsing of `scr1` stops on an unimplemented return statement at `core/pipeline/scr1_ipic.sv:96`. I will open an issue to track that support.

Thank you!

> I see your affiliation with the Russian academy of sciences. What do you intent on using this tooling for?

We have a research project on implementing static analysis methods of SystemVerilog based on `slang` and plan to use `yosys` for synthesis. Thanks to your `yosys-slang` project, now `yosys` can fully support the SystemVerilog standard.

closed 2024-08-06 by rubenbuchatskiy