povik/sv-elab #19: Support for break statements inside combinational for-loops

alikates, 2024-08-04:
Hi!
While trying to synthesize this core: https://github.com/bsc-loca/core_tile, I found out that having a break statement inside a for loop is not supported. The issue I'm seeing is inside the DCache of the design, exactly [here](https://github.com/bsc-loca/cv-hpdcache/blob/4e354ed2585257af43614ab9ac856a9bb7a1451e/rtl/src/hwpf_stride/hwpf_stride_wrapper.sv#L111-L119):
```
yosys> read_slang -f filelist_fixed.f 

1. Executing SLANG frontend.
{
  "kind": "Break"
}
Source line rtl/dcache/rtl/src/hwpf_stride/hwpf_stride_wrapper.sv:116.17-116.23:                 break;
<suppressed ~141 debug messages>
ERROR: Feature unimplemented at slang_frontend.cc:1269, see AST and code line dump above
```

I' ve looked a bit at the frontend code and saw that full cases are already supported, so I think it could be possible to transform these kind of loops into a case with priority.

For now I've given a couple hours to study the code but I'm struggling a bit figuring out where to start. Let me know if you already had this in mind. Otherwise I'd appreciate a bit of guidance on where to start to better understand the code to try add support for it.

Thanks!

alikates, 2024-08-11:
Thanks! I will contact you if I have some question about the code. Anyways I think the main issue was that I wasn't really familiar with Yosys RTLIL either way, but I'm slowly learning about both!

alikates, 2024-08-16:
I guess we can close this already, right?

povik, 2024-08-16:
Yes, let's open a separate issue to track getting core_tile through yosys-slang, as that has turned out to not be a bad codebase to test the tool on.

closed 2024-08-16 by povik