povik/sv-elab #26: Missynthesis with break statements

povik, 2024-08-13:
The break statement support doesn't take the effects of the step expression into account. We missynthesize a priority encoder if written this way:
```
    always_comb begin: for_loop_priority_encoder
        integer i = 0;
        encoded = 1'b0;
        for (i = 0; i < INPUT_WIDTH; i++) begin
            if (bits[i]) begin
                break;
            end
        end
        encoded = i;
    end
```

alikates, 2024-08-13:
The step expression is not evaluated inside the conditional branch for:
```cpp
SwitchHelper b(current_case, vstate, {RTLIL::S0});
			b.sw->statement = &stmt;

			b.branch({substitute_rvalue(disable)}, [&]() {
				current_case->statement = &stmt.body;
				eval.push_frame(nullptr, disable);
				stmt.body.visit(*this);
				eval.pop_frame();
			});
			b.finish(netlist);

			for (auto step : stmt.steps)
				eval(*step);

			ncycles++;
```
I assume that the effect of the step should also be tracked inside the branch.

povik, 2024-08-13:
Yes, but at the same time we need for it to be constant folded for the `cv` check, which may require some restructuring of the case tree to work out. (Specifically subsequent iterations need to be below earlier iterations on the case tree.) I can look into it later.

alikates, 2024-08-13:
Okay, thanks!

povik, 2024-08-15:
Should be addressed in e08d249

closed 2024-08-15 by povik