povik/sv-elab #27: Unknown module on loaded standard cells

phsauter, 2024-08-16:
Explicit instantiations of standard cells like this:
```sv
module tc_clk_inverter (
    input  logic clk_i,
    output logic clk_o
  );
  (* keep *)(* dont_touch = "true" *)
  sg13g2_inv_1 i_inv (
    .A ( clk_i ),
    .Y ( clk_o )
  );
```

Produce the following error:
```
target/ihp13/src/tc_clk.sv:14:3: error: unknown module 'sg13g2_inv_1'
  sg13g2_inv_1 i_inv (
```

Even if the standard cells have been loaded into Yosys using `read_liberty`.

My guess is that the previously loaded modules/cells are not communicated from Yosys to slang.

povik, 2024-08-16:
> My guess is that the previously loaded modules/cells are not communicated from Yosys to slang.

That's right. As a stopgap measure there's `--ignore-unknown-modules`, see also #3.

povik, 2024-09-28:
Making previously loaded modules visible to slang is now implemented as the `--extern-modules` option:

> Import as an instantiable blackbox any module which was previously loaded into the current design by a Yosys command; this allows composing hierarchy of SystemVerilog and non-SystemVerilog modules

In the future this option might become the default, but as it is now it has some rough edges (e.g. it needs to be used in conjunction with `--top` as it can confuse the top module auto-selection)

closed 2024-09-28 by povik