povik/sv-elab #28: Limited translate on/off pragma support

phsauter, 2024-08-16:
From what I can tell the translate on/off preprocessing currently does not support arbitrary markers or the most commonly used ones.

[For example in cva6](https://github.com/pulp-platform/cva6/blob/ee89dcc00e6c1a1f4cf97ee1835e950fcfdeebb5/core/cva6.sv#L1399) this does not seem to be a supported string at the moment.

Here is a pretty complete list of common [metacomments](https://insights.sigasi.com/tech/list-known-vhdl-metacomment-pragmas/) (it is for VHDL but they are similarly used in Verilog and SystemVerilog).

povik, 2024-08-16:
As it happens that's what I am working on just this moment. There are two ways in which the current support is limited:

 * The detected pragma text is fixed to `synopsys translate_on` and `_off`

 * The pragma will only be detected at a top-level module scope, and not for example inside a procedural block

I am working on having the slang preprocessor detect the pragmas (through a user-configurable regex) and skip the tokens before they are handed over to parsing. This should address both points.

povik, 2024-08-27:
> I am working on having the slang preprocessor detect the pragmas (through a user-configurable regex) and skip the tokens before they are handed over to parsing.

The final piece enabling this is pulled in as a slang patch in 7cec67fefe73f91293cbb5b45f651923aa70bc8d. For the time being the regexes are hard-coded to

    \/\/.*(synopsys|pragma).+(translate_on|synthesis_on).*
    \/\/.*(synopsys|pragma).+(translate_off|synthesis_off).*

and the feature is enabled unconditionally. Later on there might be a standard slang cli option for configuring the non-standard pragmas coming out as the result of MikePopoloski/slang#1100 or a subsequent PR.

closed 2024-08-27 by povik