povik/sv-elab #29: unable to build

astrollo, 2024-08-16:
I got an error in the final step of building yosys-slang.
My system:  Ubuntu 22.04.4 LTS
Yosys version: Yosys 0.44+9 (git sha1 4b9f45273, g++ 11.4.0-1ubuntu1~22.04 -fPIC -fPIC -O3)

Error while executing yosys-config:

yosys-config --build build/slang.so slang_frontend.cc initial_eval.cc builder.cc -Ibuild/slang_install/include -std=c++20 -DSLANG_BOOST_SINGLE_HEADER -Lbuild/slang_install/lib -lsvlang -lfmt

/usr/bin/ld: build/slang_install/lib/libfmt.a(format.cc.o): warning: relocation against `_ZTIN3fmt3v1012format_facetISt6localeEE' in read-only section `.text'

/usr/bin/ld: build/slang_install/lib/libfmt.a(format.cc.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC

/usr/bin/ld: final link failed: bad value

collect2: error: ld returned 1 exit status

povik, 2024-08-16:
Thanks, let me take a look. Is this with yosys from Ubuntu’s distribution?

astrollo, 2024-08-16:
No, I have built yosys from source (g++ 11.4.0)

povik, 2024-08-16:
So to reproduce I will install as much as I need from packages, then compile Yosys in default configuration, install it, then attempt to compile yosys-slang, is that right?

astrollo, 2024-08-16:
Yes, I did it this way

astrollo, 2024-08-20:
Solved. The README reports the following switches for cmake, that result in the ld error:

`cmake -S third_party/slang -B build/slang -DCMAKE_INSTALL_PREFIX=build/slang_install -DSLANG_USE_MIMALLOC=OFF`

Using instead all the swicthes found in the script build.sh, yosys-slang is correctly build:

```
cmake -S third_party/slang -B build/slang -DCMAKE_INSTALL_PREFIX=build/slang_install -DSLANG_USE_MIMALLOC=OFF \
        -DSLANG_INCLUDE_TESTS=OFF -DCMAKE_BUILD_TYPE=Release -DSLANG_INCLUDE_TOOLS=OFF \
        -DCMAKE_CXX_FLAGS="-fPIC" -DBoost_NO_BOOST_CMAKE=ON
```
PLease, update the README

closed 2024-08-20 by astrollo

povik, 2024-08-20:
Thanks! I worked on reproducing this but got stuck by slang throwing a compiler error when built in a Ubuntu container.

> Please, update the README

I am in the process of moving to a Makefile which will invoke the cmake step for you so those README instructions will be retired soon.