povik/sv-elab #33: Memories being incorrectly inferred

alikates, 2024-08-18:
I was checking on some of the sv-tests and noticed they were crashing the frontend.
I ran this test with GDB:
```systemverilog
module top ();

bit _bit [7:0];
logic _logic [7:0];
reg _reg [7:0];

endmodule
```
Which results in:
```
yosys> read_slang tests/selects/unpacked.sv 

1. Executing SLANG frontend.
Adding _bit (Variable)
Adding _logic (Variable)
Adding _reg (Variable)
ERROR: Assert `wire' failed in slang_frontend.cc:2645.
```
And I noticed that for all of the signals, it's inferring a memory instead of adding them as wires. This causes the assertion to fail as the wire is not found later on.

Also, for catching issues like this easier, I thought it would be nice to have some debug message to know when this occurs without having to check with a debugger.

povik, 2024-08-18:
I think it's okay to infer memories from those. Any variable with an unpacked dimension is a potential memory unless you do something to it which cannot be mapped to memory accesses. Later on in the flow memories can be converted to flip-flops, so you shouldn't be losing anything by eager inference in the frontend.

> This causes the assertion to fail as the wire is not found later on.

Yes, this happens because any `bit` variable is implicitly initialized to `'0`, but we don't support initialization data on memories yet. The memory detection accounts for it, but not if the initialization is implicit.

> Also, for catching issues like this easier, I thought it would be nice to have some debug message to know when this occurs without having to check with a debugger.

Agreed

alikates, 2024-08-18:
Aah, understood! Thanks for clarifying. So the inference means it's a potential memory, rather than directly converting it to one.

povik, 2024-08-18:
Yes, we infer an RTLIL memory which means it may or may not be mapped to a technology memory. I don't think people use implicit inference in ASIC flows (e.g. in the `synth` command which people seem to be basing ASIC flows on, there's a `memory_map` call which will convert all those implicit memories to flip-flops unconditionally). In FPGA flows you usually have a number of memory-like targets, one of which is the flip-flop fallback, and there's some cost heuristics to select a fitting target for each inferred memory (that happens in `memory_libmap`).

povik, 2024-08-20:
Addressed in 409ef8b79431a9c2a2d2d54ed6825accbf264385

closed 2024-08-20 by povik