povik/sv-elab #35: While loop unrolling should have a limit

povik, 2024-08-18:
The frontend hangs on input like
```
always_comb begin
while(1) begin end
end
```
Implement a user-configurable limit to loop unrolling.

alikates, 2024-08-19:
We can have both a loop unroll limit and also a check for when the expression is constant and always true

povik, 2024-09-25:
Added in 569799b, unrolling behavior sampled below

### Case 1

```
module top(input logic [7:0] bound);
	always_comb for (int i = 0; i < bound; i++)
		;
endmodule
```

goes through

### Case 2

```
module top2(input logic [15:0] bound);
	always_comb for (int i = 0; i < bound; i++)
		;
endmodule
```

Note the different `bound` signal size. It raises an error:

```
src/test_bounds.sv:7:14: error: unroll limit of 4000 exhausted [--unroll-limit=]
        always_comb for (int i = 0; i < bound; i++)
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

### Case 3

The unrolling of nested loops compounds and is counted together towards the unroll limit:

```
module top3();
	always_comb
		for (int i = 0; i < 100; i++)
		for (int j = 0; j < 100; j++)
		;
endmodule
```

```
src/test_bounds.sv:14:3: error: unroll limit of 4000 exhausted [--unroll-limit=]
                for (int j = 0; j < 100; j++)
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/test_bounds.sv:13:3: note: loop contributes to unroll tally
                for (int i = 0; i < 100; i++)
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

closed 2024-09-25 by povik