povik/sv-elab #43: Error with `boost_unordered.hpp` (not found)

hpretl, 2024-09-04:
I am trying to compile yosys-slang on Ubuntu 22.04 LTS with Boost compiled from sources and located in `/usr/local/boost`, and I get the following compile error:
```
In file included from build/slang_install/include/slang/text/SourceLocation.h:14,
                 from build/slang_install/include/slang/diagnostics/Diagnostics.h:18,
                 from build/slang_install/include/slang/ast/Lookup.h:11,
                 from build/slang_install/include/slang/ast/ASTContext.h:12,
                 from build/slang_install/include/slang/ast/Expression.h:10,
                 from build/slang_install/include/slang/ast/TimingControl.h:10,
                 from src/async_pattern.cc:7:
build/slang_install/include/slang/util/Hash.h:17:14: fatal error: boost_unordered.hpp: No such file or directory
   17 | #    include <boost_unordered.hpp>
      |              ^~~~~~~~~~~~~~~~~~~~~
```

Any idea what is going on?

hpretl, 2024-09-04:
Additional information:
- there is a `third_party/slang/external/boost_unordered.hpp` which seems not to be found
- my local Boost 1.82 only has these:
```
/usr/local/include/boost > ll -R | grep unordered
drwxr-xr-x  3 root root  4096 Aug 21 13:39 unordered
-rw-r--r--  1 root root   568 Aug 21 13:39 unordered_map.hpp
-rw-r--r--  1 root root   568 Aug 21 13:39 unordered_set.hpp
-rw-r--r-- 1 root root  6049 Aug 21 13:39 unordered_multiset_of.hpp
-rw-r--r-- 1 root root  5881 Aug 21 13:39 unordered_set_of.hpp
-rw-r--r-- 1 root root  8150 Aug 21 13:39 unordered_associative_container_adaptor.hpp
-rw-r--r-- 1 root root  3816 Aug 21 13:39 unordered_map_adaptor.hpp
-rw-r--r-- 1 root root  3250 Aug 21 13:39 unordered_multimap_adaptor.hpp
-rw-r--r-- 1 root root  3016 Aug 21 13:39 unordered_multiset_adaptor.hpp
-rw-r--r-- 1 root root  2811 Aug 21 13:39 unordered_set_adaptor.hpp
-rw-r--r-- 1 root root 1696 Aug 21 13:39 unordered_set_support.hpp
-rw-r--r-- 1 root root  5678 Aug 21 13:39 unordered_map_view.hpp
-rw-r--r-- 1 root root  4665 Aug 21 13:39 unordered_multimap_view.hpp
-rw-r--r-- 1 root root  2167 Aug 21 13:39 unordered_multiset_view.hpp
-rw-r--r-- 1 root root  1958 Aug 21 13:39 unordered_set_view.hpp
-rw-r--r-- 1 root root   953 Aug 21 13:39 is_unordered_range.hpp
-rw-r--r-- 1 root root 13505 Aug 21 13:39 iunordered_set_index.hpp
-rw-r--r-- 1 root root  4239 Aug 21 13:39 unordered_map_index.hpp
-rw-r--r-- 1 root root  17011 Aug 21 13:39 unordered_set_hook.hpp
-rw-r--r-- 1 root root  46437 Aug 21 13:39 unordered_set.hpp
-rw-r--r-- 1 root root   833 Aug 21 13:39 isunordered.hpp
-rw-r--r-- 1 root root 2419 Aug 21 13:39 is_unordered_set_or_map.hpp
-rw-r--r-- 1 root root  8618 Aug 21 13:39 ptr_unordered_map.hpp
-rw-r--r-- 1 root root  8831 Aug 21 13:39 ptr_unordered_set.hpp
-rw-r--r-- 1 root root  1239 Aug 21 13:39 serialize_ptr_unordered_map.hpp
-rw-r--r-- 1 root root  1207 Aug 21 13:39 serialize_ptr_unordered_set.hpp
-rw-r--r-- 1 root root 2794 Aug 21 13:39 archive_input_unordered_map.hpp
-rw-r--r-- 1 root root 2275 Aug 21 13:39 archive_input_unordered_set.hpp
-rw-r--r-- 1 root root 4272 Aug 21 13:39 boost_unordered_map.hpp
-rw-r--r-- 1 root root 4044 Aug 21 13:39 boost_unordered_set.hpp
-rw-r--r-- 1 root root 2494 Aug 21 13:39 unordered_collections_load_imp.hpp
-rw-r--r-- 1 root root 3049 Aug 21 13:39 unordered_collections_save_imp.hpp
-rw-r--r-- 1 root root 4164 Aug 21 13:39 unordered_map.hpp
-rw-r--r-- 1 root root 4024 Aug 21 13:39 unordered_set.hpp
./unordered:
-rw-r--r-- 1 root root   1716 Aug 21 13:39 unordered_flat_map_fwd.hpp
-rw-r--r-- 1 root root  27671 Aug 21 13:39 unordered_flat_map.hpp
-rw-r--r-- 1 root root   1641 Aug 21 13:39 unordered_flat_set_fwd.hpp
-rw-r--r-- 1 root root  19872 Aug 21 13:39 unordered_flat_set.hpp
-rw-r--r-- 1 root root   2819 Aug 21 13:39 unordered_map_fwd.hpp
-rw-r--r-- 1 root root 106107 Aug 21 13:39 unordered_map.hpp
-rw-r--r-- 1 root root   1716 Aug 21 13:39 unordered_node_map_fwd.hpp
-rw-r--r-- 1 root root  32285 Aug 21 13:39 unordered_node_map.hpp
-rw-r--r-- 1 root root   1641 Aug 21 13:39 unordered_node_set_fwd.hpp
-rw-r--r-- 1 root root  24164 Aug 21 13:39 unordered_node_set.hpp
-rw-r--r-- 1 root root   2619 Aug 21 13:39 unordered_set_fwd.hpp
-rw-r--r-- 1 root root  78818 Aug 21 13:39 unordered_set.hpp
```

hpretl, 2024-09-04:
BTW, I also compile `slang`, and it builds fine. Using a script like this:
```
git clone --filter=blob:none "${SLANG_REPO_URL}" "${SLANG_NAME}"
cd "${SLANG_NAME}" || exit 1
git checkout "${SLANG_REPO_COMMIT}"
cmake -B build -DSLANG_INCLUDE_TESTS=OFF
cmake --build build -j"$(nproc)"
cmake --install build --strip --prefix="${TOOLS}/${SLANG_NAME}"
```

povik, 2024-09-04:
I am purposefully trying for the slang build not to use system Boost, and to instead rely on the vendored header at `third_party/slang/external/boost_unordered.hpp`, to make linking against slang simpler.

I supply `-DBoost_NO_BOOST_CMAKE=ON` to the cmake configure step to make that happen, but I suspect it has found the `/usr/local/include/boost` installation of Boost anyway.

Is there anything in your build environment which would point build systems to `/usr/local/include/boost`?

When the yosys-slang build fails, is there `build/slang_install/include/boost_unordered.hpp`? It should be there if the slang build was configured for using the vendored header.

povik, 2024-09-04:
Maybe I should be passing `-DCMAKE_DISABLE_FIND_PACKAGE_Boost=ON` to cmake instead

povik, 2024-09-04:
Can you make the change to the Makefile, and test if that resolves your issue? On my machine Boost isn't found either way, so I can't confirm/rule out it works.

hpretl, 2024-09-04:
Looks good:
```
/tmp/yosys-slang > make
make build-slang
make[1]: Entering directory '/tmp/yosys-slang'
make configure-slang
make[2]: Entering directory '/tmp/yosys-slang'
cmake -S .//third_party/slang -B build/slang \
	-DCMAKE_INSTALL_PREFIX=build/slang_install \
	-DSLANG_INCLUDE_TESTS=OFF \
	-DSLANG_INCLUDE_TOOLS=OFF \
	-DCMAKE_BUILD_TYPE=Release \
	-DSLANG_USE_MIMALLOC=OFF \
	-DCMAKE_CXX_FLAGS="-fPIC" \
	-DCMAKE_DISABLE_FIND_PACKAGE_Boost=ON \
	-DCMAKE_DISABLE_FIND_PACKAGE_fmt=ON
-- CMake version: 3.22.1
-- slang version: 6.0.0+713164b8
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Build STATIC library as: svlang
-- {fmt} version: 11.0.2
-- Build type: Release
-- Using remote fmt library
-- Using vendored boost_unordered header
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_DISABLE_FIND_PACKAGE_fmt

-- Build files have been written to: /tmp/yosys-slang/build/slang
```

Since we are looking into this, maybe you could get rid of the warning as well?

And the build works now!

povik, 2024-09-04:
Pushed [4e034b3](https://github.com/povik/yosys-slang/commit/4e034b397b78689d9c7ee59c589509850221f62c)

povik, 2024-09-04:
There isn't a warning in the CI logs either: https://github.com/povik/yosys-slang/actions/runs/10702947984/job/29672311857

I wonder what's different in your setup

hpretl, 2024-09-04:
If you know of a way to find out, then I could test it. I am inside a Docker VM, running on macOS, but this should not make any difference in this case.

povik, 2024-09-04:
Thanks. Let me close this issue and open another one for the warning

closed 2024-09-04 by povik