povik/sv-elab #44: Remove warning from the embedded cmake step

povik, 2024-09-04:
In some build environments (but not all) the following warning appears:

```
/tmp/yosys-slang > make
make build-slang
make[1]: Entering directory '/tmp/yosys-slang'
make configure-slang
make[2]: Entering directory '/tmp/yosys-slang'
cmake -S .//third_party/slang -B build/slang \
	-DCMAKE_INSTALL_PREFIX=build/slang_install \
	-DSLANG_INCLUDE_TESTS=OFF \
	-DSLANG_INCLUDE_TOOLS=OFF \
	-DCMAKE_BUILD_TYPE=Release \
	-DSLANG_USE_MIMALLOC=OFF \
	-DCMAKE_CXX_FLAGS="-fPIC" \
	-DCMAKE_DISABLE_FIND_PACKAGE_Boost=ON \
	-DCMAKE_DISABLE_FIND_PACKAGE_fmt=ON
-- CMake version: 3.22.1
-- slang version: 6.0.0+713164b8
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Build STATIC library as: svlang
-- {fmt} version: 11.0.2
-- Build type: Release
-- Using remote fmt library
-- Using vendored boost_unordered header
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_DISABLE_FIND_PACKAGE_fmt

-- Build files have been written to: /tmp/yosys-slang/build/slang
```

Reported by @hpretl in #43.

This ticket is to either understand why it shows up, or to address it.

akashlevy, 2024-12-18:
What happens if you remove `-DCMAKE_DISABLE_FIND_PACKAGE_fmt=ON` from the Makefile on the affected platforms? Is this define even needed?

I'm not seeing any issues either with or without this flag on CentOS 7 (x64) and macOS (arm64)

povik, 2024-12-19:
> Is this define even needed?

The define is there to prevent cmake from picking up the system version of `fmt` for slang, otherwise the plugin gets improperly linked. Ideally we would adapt to either case in the plugin's Makefile but we need to find a clean solution for that.

PR #105 (2025-04-02): Convert build system to use CMake
Fixes #44.
Fixes #91.