povik/sv-elab #56: lack of warnings/errors for unwanted latches

astrollo, 2024-10-02:
Consider the following trivial (and buggy) code: 
 ```
module test1 (input logic a, b, output logic y);
always_comb begin
  if (a)
    y = b;
end
endmodule
```

`read_slang` does not give any error or warning, and a muliplexer with output connected to in_0 is synthesized. 
`read_verilog -sv` followed by: `proc` gives correctly the following message and exit: ERROR: Latch inferred for signal \test1.\y from always_comb process
Similar problems occur for `case `statement, with case values incompletely covered (no `default`)

povik, 2024-10-02:
Thanks, that needs addressing

povik, 2024-10-02:
Actually I am not sure it needs addressing, this will emit a combinational loop which will be caught in a subsequent `check` call. To me a combinational loop is not an invalid reading of the input code when `always_comb` prohibits inference of latches.

We could detect this in the frontend and emit a warning/error, but there's risk of false positives depending on how we do the detection. In any case I reclassify this as an enhancement.

phsauter, 2024-10-02:
I think the way it is currently handled in yosys-slang itself is fine (ie `check` catching the loop).  
What would be more interesting would be a warning (or just info) in Slang itself that warns of incomplete cases.  
[-Wcase-gen-dup](https://sv-lang.com/warning-ref.html#case-gen-dup) and [-Wcase-gen-none](https://sv-lang.com/warning-ref.html#case-gen-none) show that something like that is probably possible.

MikePopoloski, 2025-07-04:
FWIW slang now has warnings like you're talking about here:
* https://sv-lang.com/warning-ref.html#inferred-latch
* https://sv-lang.com/warning-ref.html#case-incomplete