povik/sv-elab #62: Memories with initialization fail parsing

povik, 2024-10-16:
Minimal test case:
```
module top();
	reg [7:0] m[2];
	initial m[0] <= 0;
endmodule
```

povik, 2024-11-21:
Fixed in [1e9c97e](https://github.com/povik/yosys-slang/commit/1e9c97e11c728d8685f92d0860b348b723b2082f), follow-up issue #69

closed 2024-11-21 by povik

tarikgraba, 2025-01-23:
Hi,

```verilog
module sram(input clk,
            input we,
            input [9:0] addr,
            input [31:0] d_i,
            output logic [31:0] d_o );

logic [31:0] mem [1024];

initial
  $readmemh("load.x",mem);

always_ff@(posedge clk) begin
  if(we)
     mem[addr] <= d_i;
  d_o <= mem[addr];
end

endmodule
```

This fails with

```
variable 'mem' driven by always_ff procedure cannot be written to by any other process
```
Does not fail with `always`.

While the standard restricts driving variable in multiple processes when always_ff is used, a lot of tool consider this code as OK (tested questa, quartus, vivado and even yosys own read_verilog -sv).