povik/sv-elab #70: Feature unimplemented (xor X[N-1:0] (a, b, c))

miquelt9, 2024-12-07:
Hi! I was willing to read the following system verilog:

```
module bus_invert(in, out, invert);
    parameter N=8;
    input [N-1:0] in;
    input invert;
    output [N-1:0] out;
    xor X[N-1:0] (out, in, invert);
endmodule
```
However it seems it is a feature not yet implemented:

```
yosys> read_slang program_x.v 

1. Executing SLANG frontend.
{
  "name": "",
  "kind": "PrimitiveInstance",
  "addr": 96420748295152,
  "primitiveType": "96420748278896 xor",
  "ports": [
    {
      "kind": "Assignment",
      "type": "logic[7:0]",
      "left": {
        "kind": "RangeSelect",
        "type": "logic[0:0]",
        "selectionKind": "Simple",
        "value": {
          "kind": "Conversion",
          "type": "logic[7:0]",
          "operand": {
            "kind": "NamedValue",
            "type": "logic[7:0]",
            "symbol": "96420748284944 out"
          }
        },
        "left": {
          "kind": "IntegerLiteral",
          "type": "int",
          "value": "0",
          "constant": "0"
        },
        "right": {
          "kind": "IntegerLiteral",
          "type": "int",
          "value": "0",
          "constant": "0"
        }
      },
      "right": {
        "kind": "EmptyArgument",
        "type": "logic"
      },
      "isNonBlocking": false
    },
    {
      "kind": "RangeSelect",
      "type": "logic[0:0]",
      "selectionKind": "Simple",
      "value": {
        "kind": "Conversion",
        "type": "logic[7:0]",
        "operand": {
          "kind": "NamedValue",
          "type": "logic[7:0]",
          "symbol": "96420748284208 in"
        }
      },
      "left": {
        "kind": "IntegerLiteral",
        "type": "int",
        "value": "0",
        "constant": "0"
      },
      "right": {
        "kind": "IntegerLiteral",
        "type": "int",
        "value": "0",
        "constant": "0"
      }
    },
    {
      "kind": "NamedValue",
      "type": "logic",
      "symbol": "96420748284576 invert"
    }
  ]
}
Source line program_x.v:7:9:     xor X[N-1:0] (out, in, invert);
<suppressed ~3 debug messages>
ERROR: Feature unimplemented at src/slang_frontend.cc:2947, see AST and code line dump above
```

My goal would be to read it in order to perform equivalence checking on it for a project I am doing. I have considered tools such as [s2v2](https://github.com/zachjs/sv2v/) but it seems it is also not yet supported, however I was able to perform a simulation on it using [iverilog](https://github.com/steveicarus/iverilog).

I appreciate any suggestions or comments.

povik, 2024-12-16:
I confirm built-in primitives are unsupported, but it's a relatively simple addition (at least for the basic types). I put it on my queue

ldoolitt, 2025-05-12:
Is this at all related to Issue #139?  That's for an instance array of user-defined modules, where this is for a language-defined primitive.  But the array part is the same.

miquelt9, 2025-05-12:
Yes, I believe so.

Not sure if these are related but gonna link them: https://github.com/zachjs/sv2v/issues/297 and https://github.com/YosysHQ/yosys/issues/4785

Feel free to close as duplicate

povik, 2025-05-23:
This should be fixed now with #149 and 1febb85.

closed 2025-05-23 by povik