povik/sv-elab #72: Error while trying to use `yosys show` with slang plugin #133

gmsanchez, 2024-12-19:
Hi,

I am helping the [APIO](https://github.com/FPGAwars/apio) developers to add SystemVerilog support in its ecosystem. Right now we are able to run some basic examples and we are getting some errors on anothers. I wonder if anyone could help or point me in some direction.

At the moment we are trying to run the [upduino32/blinky](https://github.com/FPGAwars/apio-examples/tree/master/examples/upduino31/blinky) example. After sourcing the OSS-CAD-SUITE I am running the following

```
yosys -m slang -p "synth_ice40 -top main -json _build/hardware.json" -q leds.v main.v oscilator.v
nextpnr-ice40 --up5k --package sg48 --json _build/hardware.json --asc _build/hardware.asc --report _build/hardware.pnr --pcf main.pcf -q
icepack _build/hardware.asc _build/hardware.bin
```

which works, but if you remove the `-q` flag on the first command you can see that yosys is using the `vlog2k` frontend. Now, if we run the following

```
yosys -m slang -f slang -p "synth_ice40 -top main -json _build/hardware.json" -q leds.v main.v oscilator.v
```

we get the following output

```
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.47+149 (git sha1 384c19119, clang++ 18.1.8 -fPIC -O3)

-- Parsing `leds.v' using frontend `slang' --

1. Executing SLANG frontend.
Top level design units:
    leds

leds.v:37:3: error: unknown module 'SB_RGBA_DRV'
  SB_RGBA_DRV #(
  ^~~~~~~~~~~

Build failed: 1 error, 0 warnings
ERROR: Compilation failed
```

The `SB_RGBA_DRV` module is defined in the yosys library `tools-oss-cad-suite/share/yosys/ice40/cells_sim.v`, but is says it is not found.

If I run similar commands on [edu-ciaa-fpga/and-gate-sv](https://github.com/FPGAwars/apio-examples/tree/master/examples/edu-ciaa-fpga/and-gate-sv) example, for example

```
yosys -p "synth_ice40 -top and_gate -json _build/hardware.json" and_gate.sv -m slang -p slang
```

in some part of the output I get the following

```
2. Executing SYNTH_ICE40 pass.

2.1. Executing Verilog-2005 frontend: /home/apio-dev/.apio/packages/tools-oss-cad-suite/lib/../share/yosys/ice40/cells_sim.v
Parsing Verilog input from `/home/apio-dev/.apio/packages/tools-oss-cad-suite/lib/../share/yosys/ice40/cells_sim.v' to AST representation.

```

and the SVG file of the circuit is created.

Any help?

povik, 2024-12-19:
I will think about improving the usability here, but for now you can try supplying an extra source file with the blackbox definition:

```
(* blackbox *)
module SB_RGBA_DRV(
	input CURREN,
	input RGBLEDEN,
	input RGB0PWM,
	input RGB1PWM,
	input RGB2PWM,
	output RGB0,
	output RGB1,
	output RGB2
);
parameter CURRENT_MODE = "0b0";
parameter RGB0_CURRENT = "0b000000";
parameter RGB1_CURRENT = "0b000000";
parameter RGB2_CURRENT = "0b000000";
endmodule
```

The frontend supports a couple of different scenarios with regards to blackboxes, so far those are only documented in this test: https://github.com/povik/yosys-slang/blob/master/tests/various/blackbox_scenarios.ys

This being a blackbox with parameters it restricts our options. One way to go is making sure the frontend can directly read `share/yosys/ice40/cells_sim.v`.

gmsanchez, 2024-12-19:
Thanks for the quick response.

How can I make sure that the frontend can directly read `cells_sim.v`? I tried running `read_slang` on that file before running the other commands and fails too.
 
On other simpler examples  `cells_sim.v` seems to be loaded after reading the main file, but these simpler examples do not use blackboxes.

KrystalDelusion, 2024-12-19:
@povik could you do something like the read_verilog front end `-defer` option?

KrystalDelusion, 2024-12-19:
@gmsanchez looking at https://github.com/povik/yosys-slang/blob/c71a9451752d4c514c037ca5e88d9a0047cfa66f/tests/various/blackbox_scenarios.ys#L48-L79
I think you need to load the cells_sim *before* the slang front-end, which means you can't pass the files as command line options, you would need something like `yosys -m slang -p "read_verilog -D ICE40_HX -lib -specify +/ice40/cells_sim.v; read_slang leds.v main.v oscilator.v"` (you may need to put a read_slang command for each of the files, I'm not sure if that works as is).  If that doesn't raise any errors then you should be able to run whatever commands (synth_ice40 or show).

gmsanchez, 2024-12-19:
Thanks again for the help

If I run the command you provide I get the following

```
yosys -m slang -p "read_verilog -D ICE40_HX -lib -specify +/ice40/cells_sim.v; read_slang leds.v main.v oscilator.v"

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.47+149 (git sha1 384c19119, clang++ 18.1.8 -fPIC -O3)

-- Running command `read_verilog -D ICE40_HX -lib -specify +/ice40/cells_sim.v; read_slang leds.v main.v oscilator.v' --

1. Executing Verilog-2005 frontend: /home/apio-dev/.apio/packages/tools-oss-cad-suite/lib/../share/yosys/ice40/cells_sim.v
Parsing Verilog input from `/home/apio-dev/.apio/packages/tools-oss-cad-suite/lib/../share/yosys/ice40/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\SB_IO'.
Generating RTLIL representation for module `\SB_GB_IO'.
Generating RTLIL representation for module `\SB_GB'.
Generating RTLIL representation for module `\SB_LUT4'.
Generating RTLIL representation for module `\SB_CARRY'.
Generating RTLIL representation for module `\SB_DFF'.
Generating RTLIL representation for module `\SB_DFFE'.
Generating RTLIL representation for module `\SB_DFFSR'.
Generating RTLIL representation for module `\SB_DFFR'.
Generating RTLIL representation for module `\SB_DFFSS'.
Generating RTLIL representation for module `\SB_DFFS'.
Generating RTLIL representation for module `\SB_DFFESR'.
Generating RTLIL representation for module `\SB_DFFER'.
Generating RTLIL representation for module `\SB_DFFESS'.
Generating RTLIL representation for module `\SB_DFFES'.
Generating RTLIL representation for module `\SB_DFFN'.
Generating RTLIL representation for module `\SB_DFFNE'.
Generating RTLIL representation for module `\SB_DFFNSR'.
Generating RTLIL representation for module `\SB_DFFNR'.
Generating RTLIL representation for module `\SB_DFFNSS'.
Generating RTLIL representation for module `\SB_DFFNS'.
Generating RTLIL representation for module `\SB_DFFNESR'.
Generating RTLIL representation for module `\SB_DFFNER'.
Generating RTLIL representation for module `\SB_DFFNESS'.
Generating RTLIL representation for module `\SB_DFFNES'.
Generating RTLIL representation for module `\SB_RAM40_4K'.
Generating RTLIL representation for module `\SB_RAM40_4KNR'.
Generating RTLIL representation for module `\SB_RAM40_4KNW'.
Generating RTLIL representation for module `\SB_RAM40_4KNRNW'.
Generating RTLIL representation for module `\ICESTORM_LC'.
Generating RTLIL representation for module `\SB_PLL40_CORE'.
Generating RTLIL representation for module `\SB_PLL40_PAD'.
Generating RTLIL representation for module `\SB_PLL40_2_PAD'.
Generating RTLIL representation for module `\SB_PLL40_2F_CORE'.
Generating RTLIL representation for module `\SB_PLL40_2F_PAD'.
Generating RTLIL representation for module `\SB_WARMBOOT'.
Generating RTLIL representation for module `\SB_SPRAM256KA'.
Generating RTLIL representation for module `\SB_HFOSC'.
Generating RTLIL representation for module `\SB_LFOSC'.
Generating RTLIL representation for module `\SB_RGBA_DRV'.
Generating RTLIL representation for module `\SB_LED_DRV_CUR'.
Generating RTLIL representation for module `\SB_RGB_DRV'.
Generating RTLIL representation for module `\SB_I2C'.
Generating RTLIL representation for module `\SB_SPI'.
Generating RTLIL representation for module `\SB_LEDDA_IP'.
Generating RTLIL representation for module `\SB_FILTER_50NS'.
Generating RTLIL representation for module `\SB_IO_I3C'.
Generating RTLIL representation for module `\SB_IO_OD'.
Generating RTLIL representation for module `\SB_MAC16'.
Generating RTLIL representation for module `\ICESTORM_RAM'.
Successfully finished Verilog frontend.

2. Executing SLANG frontend.
Top level design units:
    main

leds.v:38:8: error: parameter 'CURRENT_MODE' does not exist in 'SB_RGBA_DRV'
      .CURRENT_MODE("0b1"),  // "0b0" -> full current, "0b1" -> half current.
       ^
leds.v:39:8: error: parameter 'RGB0_CURRENT' does not exist in 'SB_RGBA_DRV'
      .RGB0_CURRENT("0b000001"),
       ^
leds.v:40:8: error: parameter 'RGB1_CURRENT' does not exist in 'SB_RGBA_DRV'
      .RGB1_CURRENT("0b000001"),
       ^
leds.v:41:8: error: parameter 'RGB2_CURRENT' does not exist in 'SB_RGBA_DRV'
      .RGB2_CURRENT("0b000001")
       ^

Build failed: 4 errors, 0 warnings
ERROR: Compilation failed
```

it seems to recognize the `SB_RGBA_DRV` module now, but does not recognize the parameters. These parameters can be found in `cells_sim.v`

povik, 2024-12-19:
> @povik could you do something like the read_verilog front end -defer option?

@KrystalDelusion in principle yes. I don't know yet if this means populating abstract RTLIL modules or having a hook in hierarchy similar to what Verific has.

povik, 2024-12-19:
> How can I make sure that the frontend can directly read cells_sim.v? I tried running read_slang on that file before running the other commands and fails too.

@gmsanchez I am not sure that can be made work out-of-the-box yet.

As a stopgap solution, you can do

`read_slang leds.v main.v oscilator.v ice40_blackboxes.v`

where `ice40_blackboxes.v` contains

```
(* blackbox *)
module SB_RGBA_DRV(
	input CURREN,
	input RGBLEDEN,
	input RGB0PWM,
	input RGB1PWM,
	input RGB2PWM,
	output RGB0,
	output RGB1,
	output RGB2
);
parameter CURRENT_MODE = "0b0";
parameter RGB0_CURRENT = "0b000000";
parameter RGB1_CURRENT = "0b000000";
parameter RGB2_CURRENT = "0b000000";
endmodule
```

and any other blackboxes you need the definition for. Just make sure all are marked with the `(* blackbox *)` attribute (not all in `cells_sim.v` are).

povik, 2024-12-27:
@gmsanchez I believe the workaround above will work for you. I've changed the issue to track coming up with a solution which does not involve users writing an extra file with blackbox definitions for the slang frontend.

The upside is, for your tool, you only need to write the ICE40 library once, and then you can add it to the file list every time you are reading a SystemVerilog design targetting ICE40.

To give some background: there are issues with integrating this frontend into Yosys so that it's a 1:1 replacement for the internal Verilog frontend. There's only so much that can be done in a plugin. For ASIC flows, I've worked hard to address all the usability gaps, but you are pioneering using the frontend to program FPGAs. That should work too, but it can have rough corners like this one.

chili-chips-ba, 2026-10-11:
Reading `share/yosys/xilinx/cells_sim.v` with `read_slang` fails with 106 errors, all in its `specify` blocks, see https://github.com/YosysHQ/yosys/issues/6323. We therefore had to hack it by creating `(* blackbox *)` stubs with parameters for the primitives that our design actually uses, see
- https://github.com/chili-chips-ba/openPCIE/blob/main/4.build/hw_build.openXC7/gen_xilinx_bb.py

Merging PRs #367 and fixing Issue #282 should resolve this. We are hopeful to soon seem them mainlined to remove current workarounds from RTL.

chili-chips-ba, 2026-10-24:
... with #367 now closed and #396 merged, target xilinx works👍. The real defaults (#282) is the remaining _sv-elab_ blocker.

The FDCE specify errors and CLKSWING_CFG belong to Yosys, see yosys#6323/#6322.