povik/sv-elab #73: Error with read_slang in Yosys with slang plugin

Slaats7, 2024-12-23:
Hi,

I am fairly new to Yosys, so far I have only read Verilog files. Now I would like to read SystemVerilog files with this plugin. However I cannot get a basic example like the fifo below to be read. I wonder if anyone could help or point me in some direction.
```
yosys> debug read_slang ../rtl/design/test/fifo.v --top fifo -D DEBUG

1. Executing SLANG frontend.
Adding clk (Variable)
Adding rst (Variable)
Adding wrdata_i (Variable)
Adding wr_i (Variable)
Adding rddata_o (Variable)
Adding rd_i (Variable)
Adding s_mem_flat (Variable)
Adding s_wr_ptr (Variable)
Adding s_rd_ptr (Variable)
ERROR: Exception: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
```
The file does compile in the slang explorer: https://sv-lang.com/explore/

fifo.v:
```
module fifo (
    input logic clk,
    input logic rst,
    input logic [7:0] wrdata_i,
    input logic wr_i,
    output logic [7:0] rddata_o,
    input logic rd_i
);

    logic [127:0] s_mem_flat; // DEPTH=16, WIDTH=8
    logic [3:0] s_wr_ptr, s_rd_ptr; // PTR_WIDTH=4

    always_ff @(posedge clk or negedge rst) begin
        if (!rst) begin
            s_mem_flat <= '0;
        end else if (wr_i) begin
            s_mem_flat[(s_wr_ptr * 8) +: 8] <= wrdata_i;
        end
    end

    always_ff @(posedge clk) begin
        if (rd_i) begin
            rddata_o <= s_mem_flat[(s_rd_ptr * 8) +: 8];
        end
    end
endmodule
```

Is there something we should be doing different? Any help would be great. Thanks in advance.

povik, 2024-12-23:
No, this is a clear bug in the plugin. Thanks for reporting

povik, 2024-12-25:
So far I haven't been able to reproduce, your example reads OK for me locally. Can you share how have you obtained the slang plugin, and what platform you are running on?

Slaats7, 2024-12-26:
I am using the OSS CAD Suite (oss-cad-suite-linux-x64-20241223.tgz) and running on Linux Mint 22.

povik, 2024-12-27:
I tried enabling libc++ assertions on macOS but couldn't reproduce still. I will see about reproducing on Linux.

Slaats7, 2024-12-29:
What version of Yosys are you using? I have built Yosys and this plugin from source and now the example reads OK. I did need to use an older version of Yosys (0.43+11).

povik, 2024-12-29:
That’s a good clue. I am testing against a recentish version (I don’t have the number at hand)

povik, 2024-12-29:
> I am using the OSS CAD Suite (oss-cad-suite-linux-x64-20241223.tgz) and running on Linux Mint 22.

The CAD suite should had a nightly version of Yosys, so that's in conflict with this issue only showing on older Yosys.

Slaats7, 2024-12-30:
> > I am using the OSS CAD Suite (oss-cad-suite-linux-x64-20241223.tgz) and running on Linux Mint 22.
> 
> The CAD suite should had a nightly version of Yosys, so that's in conflict with this issue only showing on older Yosys.

I don't understand your comment. The CAD suite contained Yosys 0.48+45. When building from source, I had to downgrade Yosys to version 0.43+11.

povik, 2024-12-30:
OK, thanks for the clarification

Slaats7, 2024-12-30:
Idk if its related, but building the plugin with a newer version of Yosys (0.48+45) does not work for me.
```
$ make -j$(nproc)

    CXX build/abort_helpers.o

    CXX build/async_pattern.o

    CXX build/blackboxes.o

    CXX build/builder.o

    CXX build/slang_frontend.o

    CXX build/naming.o

In file included from src/builder.cc:7:

src/slang_frontend.h:173:60: error: ‘hash_ptr_ops’ is not a member of ‘Yosys’; did you mean ‘hash_ops’?

  173 |         Yosys::dict<const ast::Scope*, std::string, Yosys::hash_ptr_ops> scopes_remap;

      |                                                            ^~~~~~~~~~~~

      |                                                            hash_ops

src/slang_frontend.h:173:72: error: template argument 3 is invalid

  173 |         Yosys::dict<const ast::Scope*, std::string, Yosys::hash_ptr_ops> scopes_remap;

      |                                                                        ^

src/slang_frontend.h:173:16: error: ‘<expression error>’ in namespace ‘Yosys’ does not name a type

  173 |         Yosys::dict<const ast::Scope*, std::string, Yosys::hash_ptr_ops> scopes_remap;

      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

povik, 2024-12-30:
That shouldn't be related (it's due to an API change from YosysHQ/yosys#4524)

povik, 2024-12-30:
Err, I missed your point about this being the same Yosys version as the CAD suite. How come it builds for the CAD suite, I wonder?

povik, 2024-12-30:
> Idk if its related, but building the plugin with a newer version of Yosys (0.48+45) does not work for me.

So this turned out to be spot on. The last successful build of yosys-slang within CAD suite is

https://github.com/YosysHQ/oss-cad-suite-build/actions/runs/12385464988/job/34571757331

and it's been failing ever since, e.g.

https://github.com/YosysHQ/oss-cad-suite-build/actions/runs/12541062019/job/34969099348

I guess that means the CAD suite packages a stale version of the plugin, which is the likely cause of the original error.

Slaats7, 2024-12-31:
Why should this problem be limited to just the CAD suite? Doesn't this mean that the plugin can't be built with Yosys versions that have this API change?

> That shouldn't be related (it's due to an API change from [YosysHQ/yosys#4524](https://github.com/YosysHQ/yosys/pull/4524))

Therefore it will only work with older Yosys versions and the suite builds fail because they use a version with that API change?

povik, 2024-12-31:
Yes, there are two issues: (1) the plugin needs to be updated for the API change (a trivial change) and (2) OSS CAD suite distributes stale builds if the build fails

povik, 2024-12-31:
I pushed [a2e91d4](https://github.com/povik/yosys-slang/commit/a2e91d4ffa969a1c1908dbc7fc84bfe64ca00edf) which might do the trick

Slaats7, 2024-12-31:
That did not work
```
/usr/local/share/yosys/include/kernel/hashlib.h:848:36: error: ‘const struct Yosys::hashlib::hash_ptr_ops’ has no member named ‘hash’
  848 |                         hash = ops.hash(key).yield() % (unsigned int)(hashtable.size());
      |                                ~~~~^~~~
```

dpetrisko, 2024-12-31:
Hi, also running into this (directed here from searching the error :)). Just curious, is the intention for yosys-slang master to build with yosys master or is there some other versioning matrix I should follow for compatibility?

povik, 2024-12-31:
It should build and work fine with the last few yosys releases, including the latest. I think I will document this and make the CI validate it. Building with yosys master is a moving target so I can't assure it.

povik, 2024-12-31:
> Building with yosys master is a moving target so I can't assure it.

This is fundamentally at odds with the nightly releasing of OSS CAD suite so something for the Yosys team to discuss.

povik, 2024-12-31:
Second attempt at fixing the build against pre-release Yosys in 1b19302

Slaats7, 2025-01-01:
That did the trick. The plugin builds successful within the CAD suite:
https://github.com/YosysHQ/oss-cad-suite-build/actions/runs/12566168170/job/35031269381

povik, 2025-01-02:
I hope this is fixed. We can re-open if there's an issue still.

closed 2025-01-02 by povik