povik/sv-elab #74: yosys-slang ignoring verilog_defaults

dpetrisko, 2025-01-06:
Hello! Not sure if this is a bug or feature request, but I'm trying to make a generic yosys-slang flow by passing in a set of variables to a synthesis script. However, it appears that yosys-slang expects me to pass in -I during the read_slang command itself. I can work around this by constructing the TCL command and executing it, but I think it would be simpler to use the verilog_defines / verilog_defaults that already exist. I know Surelog used to define another set of variables: systemverilog_defines and systemverilog_defaults, so there may be some issue with overloading that I'm unaware of. Or is there another methodology I should be using here? Thanks!

yosys.tcl
``` 
set VDEFINES "foo=3 bar=5
foreach def ${VDEFINES} {
     verilog_defines -D${def}
 }
set VINCLUDES "/path/to/include1/ /path/to/include2/"
 foreach inc ${VINCLUDES} {
     verilog_defaults -add -I${inc}
 }
set VPKG "mypkg.sv"
set VSOURCES "src1.sv src2.sv"

read_slang {*}"${VPKG} ${VSOURCES}"
```

povik, 2025-01-06:
Hi Dan,

let me make this a feature request. The defines are accessible to a plugin, the defaults are not. I will discuss with the Yosys if they prefer exposing the defaults for me, or going with separate defines/defaults commands for yosys-slang.

povik, 2025-01-06:
Having looked into it a bit more, I'm inclined towards the following: With `verilog_defines`, which is already exposed to plugins, just reuse the data for the slang frontend. For option defaults, have a new `slang_defaults` command akin to `verilog_defaults`.

To the degree that the same option exists on all `read_verilog`, `read_slang`, `verific`, we can have the `read` command bridge over the difference and provide a frontend-agnostic interface for setting defaults. There's already the following:

```
    read -incdir <directory>

Add directory to global Verilog/SystemVerilog include directories.
```

povik, 2025-01-06:
Just to note: at the moment there's no integration of the slang frontend with `read`, but it would be the way to go to get certain features.

povik, 2025-01-06:
I've implemented slang_defaults in [d686d51](https://github.com/povik/yosys-slang/commit/d686d5111a19bd725f033fd575b6072b373454f6), once the tests pass I will push to master.

With `verilog_defaults` I've found they are not exposed to plugins in a useful way yet. I will need to work on that on the Yosys side first. In any case `slang_defaults -add -D FOO=BAR` should work.

povik, 2025-01-06:
Err, verilog_defines not verilog_defaults in the above comment (I've edited it)

akashlevy, 2025-06-04:
@povik this is complete now?