povik/sv-elab #76: Modules with `assert property` fail

7FM, 2025-01-10:
A simple module like:
```
module top    
    (
        input logic clk,
        input logic rst,
        input logic in,
        output logic out
    );

always_ff @(posedge clk) out <= in;

dummy_assertion:
    assert property (@(posedge clk) 1'b1);

endmodule
```
currently fails with the following message:
```
-- Running command `read_slang --ignore-timing --top top test/top.sv' --

1. Executing SLANG frontend.
Top level design units:
    top

test/top.sv:11:1: error: unsynthesizable timing control (ignore with '--ignore-timing')
dummy_assertion:
^
``` 
The option `--ignore-timing` does not affect the outcome.

Tested against the latest commit: https://github.com/povik/yosys-slang/commit/1b1930273972f6a0246951674a23012170e520a1

povik, 2025-01-10:
Thanks! This is a bug at least to the degree that it's a confusing error message. Let me check

povik, 2025-01-10:
Let me fix the confusing error message, but as to the `assert property` syntax itself, that's part of SVA which is not supported at the moment. I've opened #77 to track

povik, 2025-02-19:
Fixed in 368be798a41d8711fbbf957285076d97b4540939, the error message now reads

```
t.sv:11:1: error: SVA unsupported (ignore all assertions with '--ignore-assertions')
dummy_assertion:
^~~~~~~~~~~~~~~~
```

and with `--ignore-assertions` it goes away