povik/sv-elab #77: SVA (SystemVerilog assertions) support

povik, 2025-01-10:
Support for SVA is missing, e.g. an assertion written in the following way is not supported as input:

```
  assert property (@(posedge clk) req |-> ##3 ack);
```

On the other hand plain assertions like
```
  always @(posedge clk) assert(/property/);
```
are supported.

povik, 2025-01-10:
The SVA code in Verific importer is one lead for getting SVA support into yosys-slang: https://github.com/YosysHQ/yosys/blob/main/frontends/verific/verificsva.cc

CheeksTheGeek, 2025-10-17:
Hey @povik would it be possible for me to work on this?
Have worked with slang a bit before, and have been interested and looking into the theory behind formal verif recently, but it's sad to see the only foss FV solution depends on a commercial tool to allow you to use SVA. And when i started looking upstream from sby this is the spot i feel is the most viable to implement SVA support. 

I've looked into the codebase and i think i can work on it right away. Will keep you updated.

Also, looking at the conversion that VerificSVA converter follows for mapping to FSM, I'll need to make sure i document the conventions used as well as what each SVA kind breaks down to. Do you think a right spot will be the discussions for this?

dehekkendekrekker, 2025-12-06:
Hey I would like to take a shot at this one. In my RTL designs I am bumping up against the limitations of the verific plugin, and the fact that it's proprietary and I assume a big chunk of my TabbyCAD license fee goes to verific, where it should be going to yosys devs annoys me. I've fixed a bug in yosys that got merged: https://github.com/YosysHQ/yosys/pull/3516 I've followed the work done by @mndstrmr and @CheeksTheGeek . @mndstrmr's seems dormant. (Correct me if I'm wrong). @CheeksTheGeek relied too much on AI to pull this off responsibly. I am old as fuck, and have been programming since forever. I do use LLM's for my work, but I understand the importance of knowing exactly what each line does, and the ramifications if you don't. Especially since I'm working on an ASIC that I can't afford to mess up because a proof passes, when it shouldn't. @povik Since it's your repo, I suppose you have some preferences on how to to contribute. Please let me know.

povik, 2025-12-07:
Sounds good. First let me find out from the yosys team the plans they have with verificsva.cc or its future replacement and then I will get back to you with suggested next steps.

> In my RTL designs I am bumping up against the limitations of the verific plugin

Can you elaborate? To synthesize the properties we might end up doing essentially the same as the verific plugin does, with similar limitations.

As it happens @tomverbeure wrote a blog post 7 years ago on what happens in the verific plugin. I believe it's up to date.
https://tomverbeure.github.io/rtl/2019/01/04/Under-the-Hood-of-Formal-Verification.html

povik, 2025-12-07:
Here's another open codebase dealing with SVA lowering: https://github.com/diffblue/hw-cbmc

dehekkendekrekker, 2025-12-17:
I have party analyzed the problem at hand. I believe in the current state of relevant codebases (slang, yosys-slang and yosys) I will take the following approach:

Fase 1:
- Analysis of the IEEE 1800-2023 standard. I have made some headway here already.
- Determine AST test paths that lead to SVA specific behaviour.

Fase 2:
1) Implement test according to SVA behaviour test path. (sv-tests is might help, though not sure how granular those tests are)
1a) Develop test that will fail to prevent false positives
2) Use sby from TabbyCAD to verify if the test works.
3) Use yosys to extract RTLIL using the verific front-end to have reference RTLIL
4) Use sby and yosys-slang to see if the tests behave expectedly
5) Dump RTLIL generated by yosys-slang
6) Use eqy to check for equivalence between verific's RTLIL and yosys-slang's.
7) If fail, implement feature / solve bug, repeat from 4.
8) If pass, update documentation, issue PR.

I've taken a look at the codebase, and though I have a generic grasp on how it works, I will have to devote more time to really "get it" in order to do 7.

mlyoung101, 2026-05-15:
Thanks :) I'll give it a try. Since #291 was written by Claude independent of mndstrmr's work, I guess what I'll need to do is try and understand how both work independently of each other and stitch them together. It'll be less so a git rebase and more so a new implementation/manual merge type operation based off the existing works, if that makes sense. Could maybe take a little while.