povik/sv-elab #78: Error reading basic decoder

dpetrisko, 2025-01-11:
Hello! I've found an apparent discrepancy in a basic synthesis. This is a minimized testcase when trying to synthesize https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_decode.sv (in a larger design).

testcase1.tcl uses read_slang whereas testcase2.tcl uses read_verilog. The output is a $shl in both, but the width in the slang version is only a single bit instead of multi-bit. The output width is corrected by 0-padding so there is no warning, but the functionality is broken.

**testcase.sv:**
```
 module testcase
  #(localparam width_gp = 8
    , localparam lg_width_gp = $clog2(width_gp)
    )
   (input logic [lg_width_gp-1:0] in
    , output logic [width_gp-1:0] out
    );

   wire [width_gp-1:0] lo;
   wire [lg_width_gp-1:0] li;
   assign li = in;
   assign lo = (width_gp) ' (1'b1 << li);
   assign out = lo;

 endmodule
```

**testcase1.tcl:**
```
 yosys -import
 plugin -i slang
 yosys -import

 read_slang \
     testcase.sv \
     -top testcase \
     --ignore-initial \
     --ignore-timing \
     --ignore-unknown-modules \
     --best-effort-hierarchy

 hierarchy -check -top testcase

 write_verilog -nostr -noattr -noexpr -nohex -nodec testcase1.out.v
```

**testcase1.out.v:**
```
 /* Generated by Yosys 0.48+57 (git sha1 d37449605, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3) */

 module testcase(in, out);
   wire _0_;
   input [2:0] in;
   wire [2:0] in;
   wire [2:0] li;
   wire [7:0] lo;
   output [7:0] out;
   wire [7:0] out;
   \$shl  #(
     .A_SIGNED(32'b00000000000000000000000000000000),
     .A_WIDTH(32'b00000000000000000000000000000001),
     .B_SIGNED(32'b00000000000000000000000000000000),
     .B_WIDTH(32'b00000000000000000000000000000011),
     .Y_WIDTH(32'b00000000000000000000000000000001)
   ) _1_ (
     .A(1'b1),
     .B(li),
     .Y(_0_)
   );
   assign li = in;
   assign lo = { 7'b0000000, _0_ };
   assign out = lo;
 endmodule
```

**testcase2.tcl:**
```
 yosys -import

 read_verilog \
     -sv \
     testcase.sv

 write_verilog -nostr -noattr -noexpr -nohex -nodec testcase2.out.v
```

**testcase2.out.v**
```
 /* Generated by Yosys 0.48+57 (git sha1 d37449605, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3) */

 module testcase(in, out);
   wire [7:0] _0_;
   input [2:0] in;
   wire [2:0] in;
   wire [2:0] li;
   wire [7:0] lo;
   output [7:0] out;
   wire [7:0] out;
   \$shl  #(
     .A_SIGNED(32'b00000000000000000000000000000000),
     .A_WIDTH(32'b00000000000000000000000000000001),
     .B_SIGNED(32'b00000000000000000000000000000000),
     .B_WIDTH(32'b00000000000000000000000000000011),
     .Y_WIDTH(32'b00000000000000000000000000001000)
   ) _1_ (
     .A(1'b1),
     .B(li),
     .Y(_0_)
   );
   assign li = in;
   assign lo = _0_;
   assign out = lo;
 endmodule
```

**diff.txt:**
```
 <   wire _0_;
 ---
 >   wire [7:0] _0_;
 16c16
 <     .Y_WIDTH(32'b00000000000000000000000000000001)
 ---
 >     .Y_WIDTH(32'b00000000000000000000000000001000)
 23c23
 <   assign lo = { 7'b0000000, _0_ };
 ---
 >   assign lo = _0_;
```

dpetrisko, 2025-01-11:
Appears to be a casting issue, as this patch produces identical output in testcase1 / testcase2):

```
< assign lo = (width_gp) ' (1'b1 << li);
--
> assign lo = (width_gp) ' (8'b1 << li);
```

I _believe_ that the constant should be promoted, but I'll check the SV spec to see if yosys is just too generous :)

povik, 2025-01-11:
Indeed!

So far I've found:

 * Verific sides with read_verilog
 * this originates from slang since it resolves the result of the shift operation to be one bit wide
 * I have confirmed this bug persists in the current version of slang (we're lagging behind with the version which is vendored inside yosys-slang)

I will make a report upstream and/or look for a fix later.

dpetrisko, 2025-01-11:
Thanks! As an additional data point, read_verilog works without the cast so presumably the LHS makes an implicit cast

povik, 2025-01-11:
> As an additional data point, read_verilog works without the cast

read_verilog and read_slang both, right? So it's only with the cast that there seems to be an issue with yosys-slang

povik, 2025-01-11:
The text in 1800-2023 about these kinds of casts on p. 140-141 is conflicting to me.

First it says

> If the casting type is a constant expression with a positive integral value, the expression in parentheses shall
be padded or truncated to the size specified. It shall be an error if the size specified is zero or negative.

but further down it goes on to say

> When changing the size, the cast shall return the value that a packed array type with a single [n-1:0] dimension would hold after being assigned the expression, where n is the cast size.

Still the second quote is more specific, and it's how other tools seem to interpret the cast, so I expect that's the right interpretation.

dpetrisko, 2025-01-11:
> read_verilog and read_slang both, right? So it's only with the cast that there seems to be an issue with yosys-slang

just read_verilog! read_slang still does 1b

povik, 2025-01-11:
> just read_verilog! read_slang still does 1b

That's not what I see. If I read:

```
 module testcase
  #(localparam width_gp = 8
    , localparam lg_width_gp = $clog2(width_gp)
    )
   (input logic [lg_width_gp-1:0] in
    , output logic [width_gp-1:0] out
    );

   wire [width_gp-1:0] lo;
   wire [lg_width_gp-1:0] li;
   assign li = in;
   assign lo = (1'b1 << li);
   assign out = lo;

 endmodule
```
and use testcase1.tcl, I get
```
/* Generated by Yosys 0.48+47 (git sha1 UNKNOWN, clang++ 19.1.5 -fPIC -O3) */

module testcase(in, out);
  wire [7:0] _0_;
  input [2:0] in;
  wire [2:0] in;
  wire [2:0] li;
  wire [7:0] lo;
  output [7:0] out;
  wire [7:0] out;
  \$shl  #(
    .A_SIGNED(32'b00000000000000000000000000000000),
    .A_WIDTH(32'b00000000000000000000000000001000),
    .B_SIGNED(32'b00000000000000000000000000000000),
    .B_WIDTH(32'b00000000000000000000000000000011),
    .Y_WIDTH(32'b00000000000000000000000000001000)
  ) _1_ (
    .A(8'b00000001),
    .B(li),
    .Y(_0_)
  );
  assign li = in;
  assign lo = _0_;
  assign out = lo;
endmodule
```

dpetrisko, 2025-01-12:
> That's not what I see. If I read:
ignore me; you're right. I was looking at stale output. Rerunning, I see correct output.

Interestingly, there is a slight difference in the netlist without the cast. read_verilog has the A_WIDTH of the shifter at width=1, whereas read_slang has A_WIDTH at width=8. Probably this will get optimized at a future constant prop pass, but flagging that the internal representation may be different between the two right after reading

```
 13c13
 <     .A_WIDTH(32'b00000000000000000000000000000001),
 ---
 >     .A_WIDTH(32'b00000000000000000000000000001000),
 18c18
 <     .A(1'b1),
 ---
 >     .A(8'b00000001),
```

closed 2025-01-21 by povik

dpetrisko, 2025-01-21:
Confirming that this fixed the testcase and my larger design. Thanks!!

povik, 2025-01-21:
Great! Happy to hear if you end up using yosys-slang for something interesting