povik/sv-elab #80: Module port wires attributes discarded and presented incorrectly (for usage with CXXRTL blackboxes)

david-sawatzke, 2025-01-21:
I'm using cxxrtl to simulate a design and would like to introduce blackboxes for some module, here an example for a Xilinx Utrascale primitive:
```
(* cxxrtl_blackbox *)
module STARTUPE3 (
    output [3:0] DI,       // Allow receiving on the D input pin
    input  [3:0] DO,       // Allows control of the D pin output
    input  [3:0] DTS,      // Allows tristate of the D pin
    input        FCSBO,    // Controls the FCS_B pin for flash access
    input        FCSBTS,   // Tristate the FCS_B pin
    (* cxxrtl_edge = "p" *)input        USRCCLKO, // User CCLK input
    (* cxxrtl_edge = "p" *)input        USRCCLKTS,// User CCLK 3-state enable input
    output       EOS,      // Active-High output signal indicating the End Of Startup

    output       CFGCLK,   // Configuration main clock output
    output       CFGMCLK,  // Configuration internal oscillator clock output
    output       PREQ,     // PROGRAM request to fabric output
    input        GSR,      // Global Set/Reset input
    input        GTS,      // Global 3-state input
    input        KEYCLEARB,// Clear AES Decrypter Key input
    input        PACK,     // PROGRAM acknowledge input
    input        USRDONEO, // User DONE pin output control
    input        USRDONETS // User DONE 3-state enable output
);
    // Does nothing

endmodule
```

Reading this file with read_verilog results in the expected attributes, especially for the two wires with the the cxxrtl_edge attribute (all the following are from the rtlil):
```
  attribute \cxxrtl_edge "p"
  attribute \src "blackboxes/primitives.v:8.41-8.49"
  wire input 6 \USRCCLKO
  attribute \cxxrtl_edge "p"
  attribute \src "blackboxes/primitives.v:9.41-9.50"
  wire input 7 \USRCCLKTS
```

 Using read_slang with the STARTUPE3 as the top module results in:
```
  attribute \cxxrtl_edge 8'01110000
  attribute \src "blackboxes/primitives.v:8.41"
  wire input 6 \USRCCLKO
  attribute \cxxrtl_edge 8'01110000
  attribute \src "blackboxes/primitives.v:9.41"
  wire input 7 \USRCCLKTS
```
which results in a parsing error inside cxxrtl for the cxxrtl_edge attribute:

ERROR: Attribute `cxxrtl_edge' of port `STARTUPE3.\USRCCLKO' is not a string with one character per bit(from here https://github.com/YosysHQ/yosys/blob/d50849ea8395b88c8211bed6aa45b7581fd960aa/backends/cxxrtl/cxxrtl_backend.cc#L2938)

It also differs from the `read_verilog` result, in that `read_verilog` per default assumes empy modules are a blackbox.

Using another module around the `cxxrtl_blackbox` module also further drops all attributes for the wires of the submodules:
```
  wire input 6 \USRCCLKO
  wire input 7 \USRCCLKTS
```

povik, 2025-01-21:
I will take a look at the wire attribute handling.

> It also differs from the read_verilog result, in that read_verilog per default assumes empy modules are a blackbox.

The read_verilog behavior can be emulated with —empty-blackboxes or similar (I am not sure at the moment, see ‘help read_slang’)

povik, 2025-01-21:
> Using read_slang with the STARTUPE3 as the top module results in:
> 
> ```
>   attribute \cxxrtl_edge 8'01110000
> ```

and

> Using another module around the `cxxrtl_blackbox` module (as with all other modules) also further drops all attributes for the wires of the submodules:
> 
> ```
>   wire input 6 \USRCCLKO
> ```

I believe I've fixed both as of [4f32cd9](https://github.com/povik/yosys-slang/commit/4f32cd9116fd7974a89744af4b29be194e3c76de). The latter case happens when the blackbox is explicitly tagged with `(* blackbox *)` (which I think is the right thing to do when working with cxxrtl blackboxes)

closed 2025-01-21 by povik

david-sawatzke, 2025-01-22:
Thank you for the fast fix, this did indeed fix it for me