povik/sv-elab #81: Compilation issues

sergeiandreyev, 2025-01-21:
Could you please check the attached [make.log](https://github.com/user-attachments/files/18491812/make.log) file and advise how to solve those errors, what could be a root cause?

System: Ubuntu 22.04 LTS
Yosys version: `0.48+70 (git sha1 85c425469, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)`

povik, 2025-01-21:
One thing I noticed is you're using a different compiler for the Yosys build and the yosys-slang build, but I don't think that should be an issue on its own. Can you also attach the output from running `yosys-config`?

sergeiandreyev, 2025-01-21:
yeah, I tried with different compilers 😅
Here you go:
```
...
Replacement args:
    --cxx         clang++
    --cxxflags    -Wall -Wextra -ggdb -I"/usr/local/share/yosys/include" \
                      -MD -MP -D_YOSYS_ -fPIC -I/usr/local/include -std=c++17 \
                      -O3 -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS \
                      -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB \
                      -I/usr/include/tcl8.6 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC \
                      -DYOSYS_ENABLE_COVER
    --linkflags   -rdynamic
    --ldflags     (alias of --linkflags)
    --libs        -lstdc++ -lm -lrt -lreadline -lffi -ldl -lz -ltcl8.6 -ltclstub8.6
    --ldlibs      (alias of --libs)
    --bindir      /usr/local/bin
    --datdir      /usr/local/share/yosys

All other args are passed through as they are.

Use --exec to call a command instead of generating output. Example usage:

  /usr/local/bin/yosys-config --exec --cxx --cxxflags --ldflags -o plugin.so -shared plugin.cc --libs
...
```

closed 2025-01-22 by povik