povik/sv-elab #83: Support always_ff on memory with initialization

povik, 2025-01-23:
> Hi,
> 
> ```verilog
> module sram(input clk,
>             input we,
>             input [9:0] addr,
>             input [31:0] d_i,
>             output logic [31:0] d_o );
> 
> logic [31:0] mem [1024];
> 
> initial
>   $readmemh("load.x",mem);
> 
> always_ff@(posedge clk) begin
>   if(we)
>      mem[addr] <= d_i;
>   d_o <= mem[addr];
> end
> 
> endmodule
> ```
> 
> This fails with
> 
> ```
> variable 'mem' driven by always_ff procedure cannot be written to by any other process
> ```
> Does not fail with `always`.
> 
> While the standard restricts driving variable in multiple processes when always_ff is used, a lot of tools consider this code as OK (tested questa, quartus, vivado and even yosys own read_verilog -sv). 

 _Originally posted by @tarikgraba in [#62](https://github.com/povik/yosys-slang/issues/62#issuecomment-2610824409)_

povik, 2025-01-23:
> While the standard restricts driving variable in multiple processes when always_ff is used, a lot of tools consider this code as OK (tested questa, quartus, vivado and even yosys own read_verilog -sv).

Yes, I think this is in scope for the frontend to support.

Btw, I don't think `$readmemh` is supported with yosys-slang, so I am not sure the given example works if always_ff is replaced with always.

tarikgraba, 2025-01-24:
Indeed, my message was more related to the frontend as I understand that initialization support is a work in progress.

povik, 2026-11-12:
Fixed now as of c46b66b8fad0ce48273cedacc25b36ffb49fa930

closed 2026-11-12 by povik