povik/sv-elab #84: support `$past` and related system functions

rroohhh, 2025-01-23:


povik, 2025-01-23:
Is $past an SVA feature, or can you use it in a procedural context everywhere?

rroohhh, 2025-01-24:
1800-2023 says at `16.9.3 Sampled value functions`:
> The use of these functions is not limited to assertion features; they may be used as expressions in procedural code as well.

concerning these system functions:
```systemverilog
$sampled ( expression ) 
$rose    ( expression [ , [ clocking_event ] ] ) 
$fell    ( expression [ , [ clocking_event ] ] ) 
$stable  ( expression [ , [ clocking_event ] ] ) 
$changed ( expression [ , [ clocking_event ] ] ) 
$past    ( expression1 [ , [ number_of_ticks ] [ , [ expression2 ] [ , [ clocking_event ] ] ] ] ) 
```

CheeksTheGeek, 2025-11-10:
Given $past is probably the main one to work on here in terms of thinking the design for it (since all other depend on it according to LRM), some questions:
- would it be okay if we put a max depth for the $past()? like I don't think there would ever be a need of $past(x, 1000). And so, would a good maximum be 256? Or is that a bad practice for shift register chaining, since someone who wants that much probably meant it?, or should we throw a warning?
- I was reading up on some caching done on other parts of the codebase. So something like caching based on `\$past$<depth>$<input_name>$d<stage>` should be fine for the shift register chaining created as a result of $past?

- do we want to support these other global clocking ones too (I'm still trying to figure out how to get explicit clocking events implemented as part of $past's 4th parameter, but just wondering what all would be expected in a single PR):
> 16.9.4 Global clocking past and future sampled value functions
> This subclause describes the system functions available for accessing the nearest past and future values of an expression as sampled by the global clock. They may be used only if global clocking is defined (see 14.14). These functions include the capability to access the sampled value at the global clock tick that immediately precedes or follows the time step at which the function is called. Sampled value is explained in 16.5.1. The following functions are provided.
> Global clocking past sampled value functions are as follows:
> - $past_gclk ( expression )
> - $rose_gclk ( expression )
> - $fell_gclk ( expression )
> - $stable_gclk ( expression )
> - $changed_gclk ( expression )
> Global clocking future sampled value functions are as follows:
> - $future_gclk ( expression )
> - $rising_gclk ( expression )
> - $falling_gclk ( expression )
> - $steady_gclk ( expression )
> - $changing_gclk ( expression )
> The behavior of the global clocking past sampled value functions can be defined using the sampled value functions as follows (the symbol means here “is equivalent by definition”):
> - $past_gclk(v) ≡ $past(v,,, @$global_clock)
> - $rose_gclk(v) ≡ $rose(v, @$global_clock)
> - $fell_gclk(v) ≡ $fell(v, @$global_clock)
> - $stable_gclk(v) ≡ $stable(v, @$global_clock)
> - $changed_gclk(v) ≡ $changed(v, @$global_clock)

One of the problems is that for some of these latter ones, I can't even find examples in non-SVA context, so I'm guessing for now, unless there's a requirement, I could just make a small PR, with $past and other simpler ones, (but with clocking event or no?) and then partial deferred SVA, and then the full global clocking ones, then SVA.

whitequark, 2025-11-10:
> The subsequent synthesis flow can figure out sharing of the logic.

Is FF merging even a part of the default Yosys synthesis flows?

whitequark, 2025-11-10:
> do we want to support these other global clocking ones too

Global clocking has a specific Yosys primitive that can be used, it's basically an instance with a single output port that has a special function in the flow.

povik, 2025-11-10:
> Is FF merging even a part of the default Yosys synthesis flows?

I think either opt_dff or opt_merge might do it.