povik/sv-elab #97: Memory inference has quadratic runtime

povik, 2025-03-14:
> @alaindargelas I tested your design with the slang plugin for yosys as described in #4601
> It works but your depth (2^25 = 33 million) is still a bit too high Yosys it seems...
> 
> Edit your top-level with:
> ```
> SyncSpRamBeNx64 #(.ADDR_WIDTH(ADDR_WIDTH),.DATA_DEPTH(1 << ADDR_WIDTH),.OUT_REGS(0),.SIM_INIT(1)) mem  (.*);
> ```
> Your Yosys script:
> ```
> read_slang dut.sv SyncSpRamBeNx64.sv --top dut
> ```
> And the command-line:
> ```
> yosys -m slang -s yosys.ys
> ```
> My observations on my laptop, with different values for parameter ADDR_WIDTH:
> ```
> ADDR_WIDTH=14  ->  0.16 sec, 34 MB RAM
> ADDR_WIDTH=15  ->  0.64 sec, 36 MB RAM
> ADDR_WIDTH=16  ->  2.70 sec, 40 MB RAM
> ```
> So there is no longer a crash due to excessive amount of RAM used.
> But each additional bit of address makes run time 4x and RAM usage +4MB
> For 25 bits you could expect `2.7 * 4^(25-16) = 707k sec = 8 days`, still an issue but a different one and maybe this one has a solution ;-)
> 
> Note that the slang plugin is by default more strict with the language, but it will give you much better error/warning/diagnose messages. 

 _Originally posted by @marzoul in [#4934](https://github.com/YosysHQ/yosys/issues/4934#issuecomment-2725081724)_

marzoul, 2025-03-16:
That looks like a promising source : https://github.com/povik/yosys-slang/blob/master/src/slang_frontend.cc#L203
all elements are inserted one by one in a vector, at the beginning, this is quadratic indeed.

Potential solutions could be : compute the overall size first, reserve, and place elements directly at the right place. Or insert elements at the end but keep track of their positions, then reorder, this should be O(n) too.

The returned object looks like it's returned by copy, this is also unneeded burden : using a variant of that function with an additional parameter for the destination vector would reduce this (but it's not the quadratic issue).

povik, 2025-03-17:
> That looks like a promising source : https://github.com/povik/yosys-slang/blob/master/src/slang_frontend.cc#L203 all elements are inserted one by one in a vector, at the beginning, this is quadratic indeed.

Thanks, indeed that's ripe for fixing. I've pushed [579014f](https://github.com/povik/yosys-slang/commit/579014f67d5f91da902c6f2136e5f6fe0953cb36) to staging.

With this `ADDR_WIDTH=25` goes through:
```
End of script. Logfile hash: d3c765efa8, CPU: user 13.87s system 0.93s, MEM: 4567.73 MB peak
Yosys 0.49 (git sha1 427b5a251bfb511528a177bfa92567b4b3ada3dd, clang++ 19.1.7 -fPIC -O3)
Time spent: 99% 2x read_slang (14 sec), 0% 1x opt_expr (0 sec), ...
```

marzoul, 2025-03-17:
Yay ! Excellent 👍

marzoul, 2025-03-17:
I wonder if RAM usage could be reduced too ? 4567 MB for 33 M lines of 64b lines lead to 136 bytes per init word.
Or said differently, 2 bytes per data bit + 8 bytes per memory line.
Certainly the size per bit could be reduced without loss of generality of a convenient 1 byte per data bit, of with more effort of 1 or 2 bits per data bit... to be analyzed if this comes from Slang or Yosys.

marzoul, 2025-03-17:
High RAM usage may also come from intermediate duplicate vectors of data while these are built and transferred from Slang to Yosys.

povik, 2025-03-17:
This likely comes from Yosys/yosys-slang. `RTLIL::Const` is a one-byte-per-design-bit representation. Note though memory and runtime efficiency is a low-priority concern unless it's something blatant like the quadratic scaling we just witnessed.

povik, 2025-03-17:
Looking at the code:

```
		std::vector<RTLIL::State> bits;
		bits.reserve(constval.getBitstreamWidth());

		auto elems = constval.elements();
		for (auto it = elems.rbegin(); it != elems.rend(); it++) {
			auto piece = convert_const(*it);
			bits.insert(bits.end(), piece.begin(), piece.end());
		}

		log_assert(bits.size() == constval.getBitstreamWidth());
		return bits; // is implicitly converted into RTLIL::Const
```
and the Const constructor in question:
```
	Const(const std::vector<RTLIL::State> &bits) : flags(RTLIL::CONST_FLAG_NONE), tag(backing_tag::bits), bits_(bits) {}
```
I expect this creates a copy of the vector. Maybe we could have Yosys have a specialized constructor which makes use of the fact the bits vector can be consumed. Right now I'm not sure if this just works with C++ or not.

The other option of operating on RTLIL::Const in the first place doesn't work since Const had API breaking changes recently and the plugin supports a range of Yosys versions.

povik, 2025-03-18:
The fix made its way into master.

closed 2025-03-18 by povik