povik/sv-elab #109: Support 'foreach' loops

suzizecat, 2025-04-04:
Hi there ! 

It looks like yosys-slang does not support the `foreach` construct (while it could be used in synthesizable (?) design, as it happened on a testcase I had).

The MWE is:

```systemverilog
module mwe_foreach (
        input logic a [4],
        output logic [3:0] b);

        always_comb begin
                foreach(a[i]) begin : gen_loop
                        b[i] = a[i];
                end
        end
endmodule
```

And the feedback is 
```console
$ yosys -m slang -p "read_slang mwe_foreach.sv"

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.51+107 (git sha1 f03b44959, clang++ 18.1.8 -fPIC -O3)

-- Running command `read_slang mwe_foreach.sv' --

1. Executing SLANG frontend.
Top level design units:
    mwe_foreach

mwe_foreach.sv:7:3: error: unsupported language feature
                foreach(a[i]) begin : gen_loop
                ^

Build failed: 1 error, 0 warnings
<suppressed ~3 debug messages>
ERROR: Compilation failed
```

This is ran with the latest oss-cad-suite as of today, not sure how to provide you a relevant plugin version number.

Best regards,
Julien FAUCHER

povik, 2025-04-04:
Thanks Julien,

> does not support the foreach construct (while it could be used in synthesizable (?) design, as it happened on a testcase I had).

Indeed there's no support, but it's similar to `for` loops, which we do support for synthesis.

> This is ran with the latest oss-cad-suite as of today, not sure how to provide you a relevant plugin version number.

Just yesterday we've merged #107, I'll add that to the issue template so that people know to use it.

suzizecat, 2025-04-04:
Indeed replacing the foreach loops by a normal for loop works as expected.
It is just that some designer on my team used foreach quite a bit, so I stumbled upon this while trying to run testcases.

Thanks for taking it into account !