povik/sv-elab #110: Crash on use of function with 'logic' as input

suzizecat, 2025-04-04:
Hi there !
Sorry for not being able to pinpoint the issue more accurately, but while chasing a weird issue with the usage of functions in synthetizable code, I came across the following:

```systemverilog
// mwe_func_in_ff.sv
function logic [8:0] my_sum(logic [7:0] a, logic [7:0] b);
    logic [8:0] out;
    out = a + b;
    return out;
endfunction;

module mwe (
	
    input logic  i_clk,
    input logic  i_rst_n,    
    input  logic [7:0] in,
    output logic [8:0] o1
);

    always_ff @( posedge i_clk or negedge i_rst_n ) begin : p_seq
        if (~i_rst_n) begin
            o1 <= 0;
        end else begin
            o1 <= my_sum(in,3);            
        end
    end    

endmodule 
```

gives:

```console
$ yosys -m slang -p "read_slang  mwe_func_in_ff.sv"

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.51+107 (git sha1 f03b44959, clang++ 18.1.8 -fPIC -O3)

-- Running command `read_slang  mwe_func_in_ff.sv' --

1. Executing SLANG frontend.
Top level design units:
    mwe

While generating the netlist content of HDL instance mwe
        of module mwe
        with parameters
        wire for symbol my_sum.a is missing
        (id $root.my_sum.a)
remapped scopes:
 (none)
<suppressed ~12 debug messages>
ERROR: Internal frontend error at src/slang_frontend.cc:3137, see details above
```

It is interesting to note that any of the following solves this issue:
 - declaring `my_sum` as `automatic` 
 - using a literal instead of a variable in the function call.

povik, 2025-04-04:
Interesting, I suppose moving the function definition into the module makes the issue goes away?

suzizecat, 2025-04-04:
Nice catch ! 
And it exposes the actual bug I wanted to report to begin with !😁

povik, 2025-05-06:
Fixed via #140

closed 2025-05-06 by povik