povik/sv-elab #126: Unable to initialize ROM memory

RiltonF, 2025-04-22:
As the title states, I am unable to load a memory with initial values, be it a rom or ram. 
Then usage of `initial $readmemh(init_file, memory)` doesn't seem to work.
I am using the following memories, [bram_sdp](https://github.com/projf/projf-explore/blob/main/lib/memory/bram_sdp.sv) and [rom_sync](https://github.com/projf/projf-explore/blob/main/lib/memory/rom_sync.sv).

I load these memories with `read_slang --allow-dup-initial-drivers bram_sdp.sv`, The extra `-allow-dup-initial-drivers` is needed because otherwise yosys throws an error that the memory is multi driven.

I had also tried adding the attributes `(* ram_style = "block", ram_init_file = "mem/init_file.mem" *)` but that didn't do anything. I didn't see anything in the logs.

I am building for the ECP5 fpga's (ulx3s board). This is how my full build looks like:
```
  $(YOSYS) -l build/yosys.log \                                                                                                                                                                                                   
    »       »       -m slang \                                                                                                                                                                                                      
    »       »       -f slang \                                                                                                                                                                                                      
    »       »       -p "read_verilog $(CLK_SOURCES) $(V_BLACKBOX)" \                                                                                                                                                                
    »       »       -p "read_slang --empty-blackboxes \                                                                                                                                                                             
    »       »       -I src/ -I src/memory -I build/clocks \                                                                                                                                                                         
    »       »       --top top --allow-dup-initial-drivers  \                                                                                                                                                                        
    »       »        $(SV_MEMORY) $(SV_SOURCES) top.sv \                                                                                                                                                                            
    »       »       --keep-hierarchy " \                                                                                                                                                                                            
    »       »       -p 'synth_ecp5 -json $(JSON)                 
```
I had also tried loading the memories with just `read -sv $(SV_MEMORIES)` but then I get an error that the memory instantiations are not found, even if I add the flags `--allow-use-before-declare --ignore-unknown-modules` it didn't work. I had even tried the `--extern-modules` flag, but that did nothing. I've tried reordering the read commands, but that also didn't get me far.

RiltonF, 2025-04-22:
After some more testing, just a basic `initial memory = '{default:'1};` works fine. So yosys is picking up the memory initial values. Just not the `$readmemh` I guess. My current workaround is the following...

[image]
Hoorah for the streaming operator! 🧙

povik, 2025-04-22:
Indeed `$readmemh` is unsupported, and what's worse it's getting silently ignored. This latter part is a bug.

> I had also tried loading the memories with just read -sv $(SV_MEMORIES) but then I get an error that the memory instantiations are not found, even if I add the flags --allow-use-before-declare [...]

You were on the right track with this workaround and I think it would have worked if (1) it was the `read_verilog -sv` command instead of `read -sv` and (2) the module instantiations didn't involve any parameters.

RiltonF, 2025-04-23:
I got the workaround now. Thanks for your help. Like you mentioned I needed put a wrapper with no parameters on my rom. I had previously gotten to this point, but quit after error that the parameters are not found. Also `read_verilog -sv` threw an error `ERROR: Process $proc$src/memory/rom_sync.sv:0$17 in module rom_sync contains sync rules, that's unsupported by the 'undriven' command.`, but loading it with just `read -sv` worked.

Workaround for those that might need it:
```
yosys -l yosys.log \
           -m slang \
           -p "read -sv rom_sync.sv my_wrapped_rom.sv" \
           -p "read_slang --ignore-unknown-modules --top top top.sv" \
           -p "synth_ecp5 -json out.json"
``` 
```
Generating RTLIL representation for module `$paramod$89e737b1363f27a2bdb11dd94290da7544e4b850\rom_sync'.
Create rom_sync with init file 'david_1bit.mem'.
``` 
Success!

Where can I find the list of unsupported constructs? @povik

povik, 2025-04-23:
> Where can I find the list of unsupported constructs? @povik 

Nowhere at the moment, but the tool should tell you (except for bugs like this one)

povik, 2025-07-04:
@max-kudinov We do emit `$meminit_v2` (see `transfer_var_init`) but supporting `$readmemh` would need handling inside `initial_eval.cc` as if it were a variable assignment. Note the `initial_eval.cc` code is meant to be retired (replaced with a solution based on `StatementVisitor`) at which point it might be easier to implement `$readmemh`. There's no ETA for this change.

max-kudinov, 2025-12-07:
@povik Just a reminder in case you've missed it. Still interested in this