povik/sv-elab #129: [bug] Infinite loops incorrect handling

likeamahoney, 2025-04-25:
Hi all!

Let's look on this simple SystemVerilog code sample:

```verilog
module m();
  initial
  begin
    forever begin
      #1;
    end
  end
endmodule
```

`yosys-slang`  is freezing in infinite loop on this till the end of RAM due to [this code](https://github.com/povik/yosys-slang/blob/0991117a91b7ae499ae9a6cb18fd1847b0f0b14d/src/initial_eval.cc#L638) 

And also this example has the same behavior on `yosys-slang` :

```verilog
module m();
  initial
  begin
    forever begin
      continue;
    end
  end
endmodule
```

First one example is reduced code sample from real open source `RISCV` core `openc910`.

So I suppose `yosys-slang` `EvalVisitor` needs a smth like `UnrollLimitTracking` from `slang_frontend.cc` to fix possible infinite loops visiting situations.

povik, 2025-04-27:
Not sure this is "incorrect" handling but I agree the frontend should catch this and raise an error. A long-standing plan is to retire the `initial_eval.cc` code and instead rely on `ProceduralVisitor` for initial blocks. Once that happens `UnrollLimitTracking` will apply, so this will get fixed.

povik, 2026-11-12:
Both test cases fixed as of d92a17e5ada38ac3bb691aaa6a6675b7bb9ce6a4

closed 2026-11-12 by povik