povik/sv-elab #134: `--ignore-timing` should ignore all delay (ex: `#5`) statements

akashlevy, 2025-05-02:
I was looking at the `sv-tests` flag addition PR you made and noticed that `always #5 a = ~a;` failed, even with `--ignore-timing`. I think it would be good to have `--ignore-timing` flag ignore these, instead of failing with "error: unsynthesizable feature". Below is `9.2.2.1--always.sv.log` reproduced for reference:

```
name: always
description: always check
tags: 9.2.2.1 9.4.1
files: /root/sv-tests/sv-tests/tests/chapter-9/9.2.2.1--always.sv
incdirs: /root/sv-tests/sv-tests/tests/chapter-9
top_module: 
timeout: 30
type: parsing elaboration
should_fail: 0
should_fail_because: 
defines: 
compatible-runners: all
unsynthesizable: 0
results_group: 
mode: elaboration
rc: 1
tool_success: 0
runner: yosys-slang
runner_url: https://github.com/povik/yosys-slang/tree/e23228f4b4ec5ec0b4b17f1f675debfa29458665
time_elapsed: 0.008529
user_time: 0.008529
system_time: 0.0
ram_usage: 22572
date_completed: 2025-04-30 13:53:40

yosys-slang -s /root/sv-tests/sv-tests/out/tmp/tmpelc_phew/yosys-script

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.50+56 (git sha1 9f7cdd4bd, g++ 11.4.0-1ubuntu1~22.04 -fPIC -O3)

-- Executing script file `/root/sv-tests/sv-tests/out/tmp/tmpelc_phew/yosys-script' --

1. Executing SLANG frontend.
../../../tests/chapter-9/9.2.2.1--always.sv:17:9: error: unsynthesizable feature
        always #5 a = ~a;
               ^~
ERROR: Compilation failed
```

Test case (`9.2.2.1--always.sv`):
```
// Copyright (C) 2019-2021  The SymbiFlow Authors.
//
// Use of this source code is governed by a ISC-style
// license that can be found in the LICENSE file or at
// https://opensource.org/licenses/ISC
//
// SPDX-License-Identifier: ISC

/*
:name: specparam
:description: specparam tests
:tags: 6.20.5
*/
module top();
	specparam delay = 50;
endmodule
```

akashlevy, 2025-05-02:
This problem also appears with deferred assertions (SV LRM 16.9). Example (`16.2--assert0.sv`):

```
// Copyright (C) 2019-2021  The SymbiFlow Authors.
//
// Use of this source code is governed by a ISC-style
// license that can be found in the LICENSE file or at
// https://opensource.org/licenses/ISC
//
// SPDX-License-Identifier: ISC

/*
:name: assert0_test
:description: assert #0 test
:tags: 16.4
*/
module top();

logic a = 1;

assert #0 (a != 0);

endmodule
```

povik, 2025-05-06:
Fixed in 72cae17fe813670550f52fdd0f1e2546d1eb75a2

closed 2025-05-06 by povik