povik/sv-elab #143: What's does the future hold for `inout` ports?

ldoolitt, 2025-05-07:
As much as I try to push vendor-dependent code to the edge of my designs, inout ports (like i2c SDA pins) sometimes leak into nominally portable code.  That means I want to crunch them with yosys and now yosys-slang.

Right now, yosys read_verilog does OK, but does emit warnings, like
```
Warning: Yosys has only limited support for tri-state logic at the moment. (lb_marble_slave.v:300)
```
 yosys-slang treats them as errors.
```
lb_marble_slave.v:59:14: error: direction 'InOut' on inlined port connection unsupported
        inout [3:0] twi_scl,
```
Is this situation permanent?

akashlevy, 2025-05-08:
@povik Is this on `slang` side?

akashlevy, 2025-05-08:
Confirming that this is on `yosys-slang` side: https://github.com/MikePopoloski/slang/discussions/1236

povik, 2025-05-08:
@ldoolitt 

> Is this situation permanent?

Likely not. In the meantime you can work around this limitation by using the `--keep-hierarchy` mode.

ldoolitt, 2025-05-08:
Our primary use case is three-state drivers -- written in portable Verilog -- that are routed through levels of hierarchy to the edge of the design.  As in, a top-level inout port.

There are also cases where we are lazy/efficient.  Two modules communicate, where the direction of drive is set by preprocessor or generate statements.  In this case, the wires in between are labeled inout rather than try to select between input and output with the preprocessor.  And that's impossible within standard Verilog when the direction is set by a parameter.

akashlevy, 2025-05-08:
@povik What's the problem with just setting both `wire.port_input` and `wire.port_output` in RTLIL?

povik, 2025-05-08:
You can do that if this is an RTLIL port (that's what happens in the `--keep-hierarchy` mode), but this ticket is for inline connections when the frontend is flattening hierarchy. A related issue is that the `--keep-hierarchy` mode should probably become the default.

akashlevy, 2025-05-08:
Ok got that, but I'm still confused... perhaps I'm misunderstanding something fundamental.

If it's just a wire after flattening, why do we care about the direction? SV wires don't have a direction.

akashlevy, 2025-05-08:
Is it just that it will fail `check` or something?

povik, 2025-05-08:
We could substitute the outside wire for every use of the port on the inside, if we're flattening and this is a tri-state line, but that's a special case which needs to be implemented.