povik/sv-elab #144: Issue building in docker

drewranck, 2025-05-07:
This could be total user error on my part in a Dockerfile, but wondering if anyone has built this in docker (as su) or seen a similar error?

I can't seem to reproduce this outside of Docker on the same system (ubuntu jammy or noble):

```
[ 98%] Building CXX object src/CMakeFiles/yosys-slang.dir/slang_frontend.cc.o
/app/yosys-slang/src/slang_frontend.cc: In function 'Yosys::hashlib::pool<slang_frontend::VariableBit> slang_frontend::detect_possibly_unassigned_subset(Yosys::hashlib::pool<slang_frontend::VariableBit>&, slang_frontend::Case*, int)':
/app/yosys-slang/src/slang_frontend.cc:234:25: error: 'log_debug' was not declared in this scope; did you mean 'Yosys::log_debug'?
  234 |                         log_debug("%saction %s<=%s (mask %s)\n", std::string(level, ' ').c_str(),
      |                         ^~~~~~~~~
      |                         Yosys::log_debug
In file included from /usr/local/share/yosys/include/kernel/bitpattern.h:23,
                 from /app/yosys-slang/src/slang_frontend.cc:21:
/usr/local/share/yosys/include/kernel/log.h:151:20: note: 'Yosys::log_debug' declared here
  151 | static inline void log_debug(const char *format, ...) { if (ys_debug(1)) { va_list args; va_start(args, format); logv(format, args); va_end(args); } }
      | 
```

This comes after successfully building slang and yosys separately, checking out yosys-slang:

```
root@206daaccedf6:/app# mkdir yosys-slang

root@206daaccedf6:/app# git clone --recursive https://github.com/povik/yosys-slang.git

Cloning into 'yosys-slang'...
remote: Enumerating objects: 2017, done.
remote: Counting objects: 100% (629/629), done.
remote: Compressing objects: 100% (165/165), done.
remote: Total 2017 (delta 513), reused 533 (delta 462), pack-reused 1388 (from 2)
Receiving objects: 100% (2017/2017), 528.92 KiB | 4.33 MiB/s, done.
Resolving deltas: 100% (1311/1311), done.
Submodule 'third_party/fmt' (https://github.com/fmtlib/fmt) registered for path 'third_party/fmt'
Submodule 'third_party/slang' (https://github.com/MikePopoloski/slang.git) registered for path 'third_party/slang'
Cloning into '/app/yosys-slang/third_party/fmt'...
remote: Enumerating objects: 39556, done.        
remote: Counting objects: 100% (71/71), done.        
remote: Compressing objects: 100% (48/48), done.        
remote: Total 39556 (delta 30), reused 23 (delta 23), pack-reused 39485 (from 3)        
Receiving objects: 100% (39556/39556), 16.20 MiB | 10.25 MiB/s, done.
Resolving deltas: 100% (26696/26696), done.
Cloning into '/app/yosys-slang/third_party/slang'...
remote: Enumerating objects: 61802, done.        
remote: Counting objects: 100% (6926/6926), done.        
remote: Compressing objects: 100% (403/403), done.        
remote: Total 61802 (delta 6651), reused 6533 (delta 6519), pack-reused 54876 (from 1)        
Receiving objects: 100% (61802/61802), 31.98 MiB | 10.42 MiB/s, done.
Resolving deltas: 100% (48657/48657), done.
Submodule path 'third_party/fmt': checked out '40626af88bd7df9a5fb80be7b25ac85b122d6c21'
Submodule path 'third_party/slang': checked out '303d65e8dd4e2fe45e19054b9a7291d9512a7293'

root@206daaccedf6:/app# cd yosys-slang/

root@206daaccedf6:/app/yosys-slang# ls
CMakeLists.txt  LICENSE  Makefile  README.md  build  build.sh  cmake  docs  src  tests  third_party

root@206daaccedf6:/app/yosys-slang# make

cmake -S . -B build -DYOSYS_CONFIG=yosys-config -DCMAKE_BUILD_TYPE=Release
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- slang version: 8.0.220+303d65e8
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Build STATIC library as: svlang
-- {fmt} version: 11.2.0
-- Build type: Release
-- Using remote fmt library
-- Using vendored boost_unordered header
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Using yosys: yosys-config
-- yosys-config --datdir: /usr/local/share/yosys
-- yosys-config --cxxflags (filtered): -I/usr/local/share/yosys/include;-D_YOSYS_;-I/usr/local/include;-DYOSYS_VER="0.53+3";-DYOSYS_MAJOR=0;-DYOSYS_MINOR=53;-DYOSYS_COMMIT=3;-DYOSYS_ENABLE_READLINE;-DYOSYS_ENABLE_PLUGINS;-DYOSYS_ENABLE_GLOB;-DYOSYS_ENABLE_ZLIB;-I/usr/include/tcl8.6;-DYOSYS_ENABLE_TCL;-DYOSYS_ENABLE_ABC;-DYOSYS_ENABLE_COVER
-- Got YOSYS_SLANG_REVISION: cf0e37b33daeeeb08e2913c908cdd4326c0c0f9e
-- Got SLANG_REVISION: 303d65e8dd4e2fe45e19054b9a7291d9512a7293
-- Configuring done
-- Generating done
-- Build files have been written to: /app/yosys-slang/build
cmake --build build
gmake[1]: Entering directory '/app/yosys-slang/build'
gmake[2]: Entering directory '/app/yosys-slang/build'
gmake[3]: Entering directory '/app/yosys-slang/build'
gmake[3]: Leaving directory '/app/yosys-slang/build'
gmake[3]: Entering directory '/app/yosys-slang/build'
[  0%] Building CXX object _deps/fmt-build/CMakeFiles/fmt.dir/src/format.cc.o
[  1%] Building CXX object _deps/fmt-build/CMakeFiles/fmt.dir/src/os.cc.o

```

drewranck, 2025-05-07:
What my Dockerfile looks like, if that helps.

```
FROM ubuntu:latest

ENV DEBIAN_FRONTEND=noninteractive

# Various prereqs: Verilator, Slang, Yosys

RUN apt-get update && apt-get upgrade < /dev/null

RUN apt-get install -y cmake build-essential python3 libfmt-dev git \
    python3-venv python3-pip \
    help2man perl make autoconf g++ flex bison ccache \
    libgoogle-perftools-dev numactl perl-doc < /dev/null

RUN apt-get install -y libfl2 libfl-dev zlib1g zlib1g-dev gperf \
    libreadline-dev gawk tcl-dev libffi-dev graphviz \
    xdot pkg-config libboost-system-dev \
    libboost-python-dev libboost-filesystem-dev < /dev/null

RUN apt-get install -y clang libclang1 libclang-dev libclang-cpp-dev \
  eigensoft libeigen3-dev libeigenbase-farrago-java libeigenbase-resgen-java \
  tcl tclreadline swig < /dev/null

RUN mkdir /app

# Slang
RUN mkdir /app/slang
WORKDIR /app
RUN git clone https://github.com/MikePopoloski/slang.git
WORKDIR /app/slang
RUN cmake -B build
RUN cmake --build build -j

# Yosys
RUN mkdir /app/yosys
WORKDIR /app
RUN git clone --recursive https://github.com/YosysHQ/yosys
WORKDIR /app/yosys
RUN make config-gcc
RUN make
RUN make install

# slang-yosys
RUN mkdir /app/yosys-slang
WORKDIR /app
RUN git clone --recursive https://github.com/povik/yosys-slang.git
WORKDIR /app/yosys-slang
RUN make
RUN make install

ENV PATH="/app/slang/build/bin:${PATH}"
```

drewranck, 2025-05-07:
I'll try to narrow this down with a more minimal number of things apt-get install'ed

povik, 2025-05-08:
This looks to be unrelated to Docker but caused by recent unreleased changes in Yosys. You can try building against the last release (v0.53 from two days ago) or wait on #145.

povik, 2025-05-08:
Please reopen if issue persists

closed 2025-05-08 by povik