povik/sv-elab #146: Crash on `interconnect` connection

povik, 2025-05-09:
Following input
```
module top();
	interconnect bus;
	m mi(bus);
endmodule

module m(input x);
endmodule
```
causes
```
-- Running command `read_slang interconnect.sv' --

1. Executing SLANG frontend.
Top level design units:
    top

<suppressed ~1 debug messages>
ERROR: Assert `internal_signal.bitwidth() == signal.size()' failed in slang_frontend.cc:2125.
```

akashlevy, 2025-05-15:
I did some digging on this issue.

It seems that the generic (untyped) `interconnect` type needs to have its bit width resolved dynamically as it can take on different data types which may have different bit widths, e.g., `integer`, `logic`.

Slang doesn't seem to have a way to resolve the bit width of a port dynamically depending on the data type connected, it always resolves to 0. I'm filing an issue upstream to `slang` for this.

povik, 2025-05-15:
Thanks. What's the rule for resolving the bit width? Is it assumed all connnections to the same `interconnect` instance have the same type, or is there some arbitrage in a mixed-type scenario?

akashlevy, 2025-05-15:
Yeah it gets super messy because different instances can take on different bit widths depending on what's connected. At least that's my interpretation of the LRM.

akashlevy, 2025-05-15:
I'm not sure what is supposed to happen if the bit widths are mismatched on either side of the interconnect

akashlevy, 2025-05-15:
BTW this actually does get caught as an error with this line because `isFixedSize()` evaluates to false:
```
	if (!(expr.type->isFixedSize() || expr.type->isVoid())) {
		auto &diag = netlist.add_diag(diag::FixedSizeRequired, expr.sourceRange);
		diag << expr.type->toString();
		goto error;
	}
```
But the error handler doesn't raise the issue, it just creates an `x`.
```
error:
	ret = RTLIL::SigSpec(RTLIL::Sx, (int) expr.type->getBitstreamWidth());
```
Properly supporting `interconnect` should overall be quite low priority since it is so rarely used. But might be worthwhile to at least improve the error handling?

closed 2025-05-15 by povik