povik/sv-elab #150: String parameters are converted to binary

max-kudinov, 2025-05-18:
I'm trying to make yosys-slang work with FPGA toolchain (Apicula). I've passed to `read_slang` all blackbox declarations as was suggested [here](https://github.com/povik/yosys-slang/issues/72#issuecomment-2555761913).

Synthesis and PnR passes, but on on bitstream generation I get the following:

```
The 11001000110111.0MHz frequency is outside the permissible range of 3-500MHz.
The 11001000110111.0MHz frequency is outside the permissible range of 3-500MHz.
The 11001000110111.0MHz frequency is outside the permissible range of 3-500MHz.
The 11001000110111.0MHz frequency is outside the permissible range of 3-500MHz.
```

In Verilog source PLL instance look like this:

```
rPLL #(
    .FCLKIN    ( "27" ),
    .IDIV_SEL  ( 6    ),
    .FBDIV_SEL ( 25   ),
    .ODIV_SEL  ( 8    )
) rpll_inst (
    .CLKIN   ( clk_i      ),
    .CLKOUT  ( serial_clk ),
    .LOCK    ( pll_lock   ),
    .RESET   ( '0         ),
    .RESET_P ( '0         ),
    .CLKFB   ( '0         ),
    .FBDSEL  ( '0         ),
    .IDSEL   ( '0         ),
    .ODSEL   ( '0         ),
    .PSDA    ( '0         ),
    .DUTYDA  ( '0         ),
    .FDLY    ( '0         )
);
```

Then if I do simple `yosys -m slang -p "read_slang test.sv rpll.sv; write_rtlil out.rtlil` I can see that the parameter value of `FCLKIN` is converted to binary representation `parameter \FCLKIN 16'0011001000110111`.

If run with regular `yosys -p "read_verilog -sv test.sv rpll.sv; write_rtlil out.rtlil` the string is left untouched `parameter \FCLKIN "27"`.

povik, 2025-05-19:
> look at the default value for the parameter, and if it's a string literal, set CONST_FLAG_STRING on the parameter value in the instantiation.

I have implemented this in [d6f286e](https://github.com/povik/yosys-slang/commit/d6f286e473128da2c16defae9f9bd7fc5623b76d). Please see if it fixes your use case.

max-kudinov, 2025-05-20:
> I have implemented this in [d6f286e](https://github.com/povik/yosys-slang/commit/d6f286e473128da2c16defae9f9bd7fc5623b76d). Please see if it fixes your use case.

Yes, this fixes the issue for me. Thank you for a quick fix!

After that I found another problem with parameters, created another issue #155

closed 2025-05-20 by max-kudinov