povik/sv-elab #153: Perfectly valid combinational ROM triggers warning

ldoolitt, 2025-05-20:
Production Verilog code is attached (with the usual renaming of .v to .txt).
[ds1822_driver.txt](https://github.com/user-attachments/files/20320233/ds1822_driver.txt)
It encodes a couple of very small combinational ROMs with (stylized)
```
reg [3:0] result;
always @(addr) case (addr)
  0: result=3;
  1: result=1;
  default: result=5;
endcase
```
Running this through yosys-slang yields a (non-suppressible?)
```
warning: non-edge sensitivity on a signal will be synthesized as @* sensitivity
```

While Verilog textbooks today typically do recommend writing this as
```
always @(*) case (addr)
```
the old way (a) isn't wrong, and (b) IMHO better captures the intended sensitivity.

Possibilities:
* Talk me into writing `always @(*)`
* Add (or explain) a way for yosys-slang to suppress the warning
* Best of all, don't complain _unless_ some logical dependency was missed

That is,
```
reg [3:0] result;
always @(addr) case (addr)
  0: result=3;
  1: result=x;
  default: result=5;
endcase
```
should emit a very loud warning, or even error, in the case that `x` is not a synthesis-time constant.

P.S., yosys' read_verilog does _not_ complain about constructions in the provided production code.

povik, 2025-05-20:
> the old way (a) isn't wrong, and (b) IMHO better captures the intended sensitivity.

I agree, it's not wrong. The frontend doesn't have a check for all functional dependencies being in the sensitivity list so it issues a blanket warning. Adding the check is in scope but it's of low priority since SV has `always_ff`, `always_latch`, `always_comb`.

The warning isn't suppressible but it's in scope to add an option to suppress yosys-slang warnings.

> P.S., yosys' read_verilog does not complain about constructions in the provided production code.

I just tried out and it seems to have this:
```
Note: Assuming pure combinatorial block at ff.v:3.1-7.8 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
```

povik, 2025-05-20:
I'll use this ticket to track adding the dependency check, for suppressing warnings I've opened #154.