povik/sv-elab #156: `used before its declaration` generates errors despite using `--allow-use-before-declare`

gadfort, 2025-05-21:
When using `--allow-use-before-declare`, I would assume that the design would read in correctly, but it's still generating errors:

See testcase:
[test.tar.gz](https://github.com/user-attachments/files/20373541/test.tar.gz)

To run:
```
tar xvf test.tar.gz
yosys run.ys
```

Output:
```
...
swerv.sv:10907:17: note: declared here
   logic        dec_tlu_i0_exc_valid_wb1, dec_tlu_i1_exc_valid_wb1;
                ^
  in instance: swerv.dec
swerv.sv:10877:21: error: identifier 'dec_tlu_i1_exc_valid_wb1' used before its declaration
   dec_tlu_ctl tlu (.*);
                    ^~
swerv.sv:10907:43: note: declared here
   logic        dec_tlu_i0_exc_valid_wb1, dec_tlu_i1_exc_valid_wb1;
                                          ^
  in instance: swerv.dec
swerv.sv:10877:21: error: identifier 'dec_tlu_int_valid_wb1' used before its declaration
   dec_tlu_ctl tlu (.*);
                    ^~
swerv.sv:10904:55: note: declared here
   logic dec_tlu_i1_valid_wb1, dec_tlu_i0_valid_wb1,  dec_tlu_int_valid_wb1;
                                                      ^
fatal error: too many errors emitted, stopping now [--error-limit=]

Build failed: 20 errors, 0 warnings
ERROR: Compilation failed
```

povik, 2025-05-21:
Does this reproduce with slang proper? If yes, we should report upstream.

gadfort, 2025-05-21:
@povik I just checked and yes it is also present in slang: `slang version 8.0.231+ed2e5ad2`

povik, 2025-05-21:
Reported upstream

closed 2025-05-22 by povik

povik, 2025-05-24:
Yes, we should get a submodule bump soon. I'll post here when that happens

povik, 2025-05-27:
Update merged