povik/sv-elab #160: error: condition cannot be matched to any signal from the event list

gadfort, 2025-05-27:
I'm getting this error:
```
inputs/dv.sv:345:13: error: condition cannot be matched to any signal from the event list
            if(~hwif_in.hard_reset_b) begin
inputs/dv.sv:344:20: note: asynchronous load pattern implied by edge sensitivity on multiple signals
        always_ff @(posedge clk or negedge hwif_in.hard_reset_b) begin
```

It appears that the signal is in the sensitivity list. I checked and slang does not report this error.

To replicate:
https://drive.google.com/file/d/1ZBM1rIpnyMp3Rtbk75stm44j-QgowBVV/view?usp=sharing
```
tar xvf sc_issue_caliptra-datavault_job0_asap7_asap7sc7p5t_rvt_syn0_20250527-185341.tar.gz
cd sc_issue_caliptra-datavault_job0_asap7_asap7sc7p5t_rvt_syn0_20250527-185341
./run.sh
```

povik, 2025-05-27:
Thanks for the report, looks to be an error on yosys-slang part

povik, 2025-05-28:
Minimized example:
```
module r19(input clk, input rst, input x, output q);
	typedef struct packed {
		logic x;
		logic q;
		logic clk;
		logic rst;
	} foo_t;
	foo_t bar;

	assign bar.clk = clk;
	assign bar.rst = rst;
	assign bar.x = x;
	assign q = bar.q;

	always_ff @(posedge bar.clk or negedge bar.rst) begin
		if (~bar.rst)
			bar.q <= 0;
		else
			bar.q <= bar.x;
	end
endmodule
```
raises error
```
regress.sv:172:3: error: condition cannot be matched to any signal from the event list
                if (~bar.rst)
                ^~~~~~~~~~~~~
regress.sv:171:13: note: asynchronous load pattern implied by edge sensitivity on multiple signals
        always_ff @(posedge bar.clk or negedge bar.rst) begin
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```