povik/sv-elab #170: BUG: escaped identifiers pose problems

akashlevy, 2025-06-03:
See this test case, which is valid Verilog:
```
// There must be white space after the
// string which uses escape character
module \1dff (
q,      // Q output
\q~ ,   // Q_out output
d,      // D input
cl$k,   // CLOCK input
\reset* // Reset input
);

input d, cl$k, \reset* ;
output q, \q~ ;  

endmodule
```

Error is:
```
ERROR: Found control character or space (0x20) in string '\1dff$\1dff ' which is not allowed in RTLIL identifiers
```

I think the whitespace needs to represented differently for the `IdString`.

akashlevy, 2025-06-03:
Yes, the issue is that RTLIL doesn't like that. Fortunately, if the identifier is escaped you can infer that there's a space at the end anyways, which is what `write_verilog` does.

povik, 2025-06-03:
> So just need to trim that ending space I believe

Not really, we can have escaped identifiers earlier in the path. E.g. static variable `var` inside a function with name `\1f` will be (I think) `\1f .var`. Also when the frontend flattens the instance name will prefix all identifiers coming from the instance.

omixann, 2026-03-18:
I found this same issue when trying Slang to compile the Verilog code Spade (an HDL using the language as a backend) produces. Given almost all names are preceded by `\` to handle name mangling, Spade-generated Verilog cannot work when using the Slang plugin.