povik/sv-elab #178: Support for built-in system tasks and functions

akashlevy, 2025-06-03:
This can probably be broken into sub-tasks, but built-in system tasks and functions should be supported to a reasonable extent. A good reference for the extent that it should be supported: https://docs.amd.com/r/en-US/ug901-vivado-synthesis/Verilog-System-Tasks-and-Functions

akashlevy, 2025-06-03:
Trignometric functions should probably be supported for constants:

- $asin
- $acos
- $atan
- $atan2
- $sinh
- $cosh
- $tanh 
- $sin
- $asinh
- $cos
- $ascosh
- $tan
- $ln
- $atanh
- $log10
- $exp
- $sqrt
- $hypot
- $pow: is this already supported? It has an RTLIL equivalent

CheeksTheGeek, 2025-10-22:
do you think it would be wise to just return 0 for `$isunknown()`?