povik/sv-elab #184: Support for UDP

akashlevy, 2025-06-03:
User-defined primitives (`primitive` and `table`) are not supported and are synthesizable.

povik, 2025-06-03:
One data point from the PULP group: They say UDPs are usually used to construct models and don't see a use case to synthesize them.

povik, 2025-06-03:
There are two ways UDP instances can be supported:

 * emit an instanced blackbox (for synthesis when UDP describes a technology primitive)
 * synthesize the behavioral description of the UDP (for verification)

akashlevy, 2025-06-03:
Yes, they are mainly used for PDK primitives. We have a few customers who use their PDK's `simlib.v` instead of Liberty files for speed. Verific supports UDP, so this can give some minor speedups. But we can just ask these customers to switch to Liberty files for blackboxing. So low priority.