povik/sv-elab #195: Support modules with import but no ports

sifferman, 2025-06-17:
Hello and thanks for this tool! I wanted to report a bug:

I am using this OSS CAD Suite release: <https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2025-06-17/oss-cad-suite-linux-x64-20250617.tgz>

## RTL

```systemverilog
// a.sv

package pkg;
endpackage

module a;
endmodule

module b import pkg::*;;
endmodule
```

## Passes Verilator Linting

```bash
# Passes
verilator a.sv --lint-only -Wno-MULTITOP
```

## Fails `read_slang`

```bash
# Doesn't work
yosys -p 'tcl yosys.tcl'
```
```tcl
# yosys.tcl

yosys -import
plugin -i slang
yosys -import

read_slang a.sv \
 --std latest
```
```
1. Executing SLANG frontend.
Top level design units:
    a
    b

a.sv:8:24: error: expected parameter list or port list
module b import pkg::*;;
                       ^

Build failed: 1 error, 0 warnings
ERROR: Compilation failed
```

sifferman, 2025-06-18:
I just checked, and it is also supported by VCS, Xcelium, and Genus.

From the spec (1800-2023 A.1.2):

> ```
> module_ansi_header ::=
>     { attribute_instance } module_keyword [ lifetime ] module_identifier
>         { package_import_declaration } 1 [ parameter_port_list ] [ list_of_port_declarations ] ;
> ```

I believe that it is valid syntax.

MikePopoloski, 2025-06-18:
The footnote was not in 1800-2005 because package imports weren't allowed in the header at all in 2005. The ability to import from the header was added in 2009, along with the footnote at the same time -- here is the Accellera issue on the topic: https://accellera.mantishub.io/view.php?id=329

I guess some vendors simply ignored the rule. It looks like VCS warns, Xcelium and Questa allow it without warning, and Aldec reports an error.

I guess slang can add a compatibility flag for this, or make it a downgrade-able error instead, but it seems kind of silly. Like you can just move the semicolon over and then have actual legal code:
```
module b import pkg::*;;
endmodule

--> should be

module b; import pkg::*;
endmodule
```

povik, 2025-06-18:
@MikePopoloski do you want to transfer this issue to slang proper as a feature request? Otherwise we can close.

MikePopoloski, 2025-06-18:
I'm generally happy to add compat flags if there's a genuine need, like you have a large old codebase that uses it all over and it's hard to fix. It doesn't sound like that's the case though and I don't want to end up with zillions of flags that nobody uses, so maybe let's just close it out?

closed 2025-06-19 by povik