povik/sv-elab #212: Incorrect handling of `wor`

toby-bro, 2025-07-18:
When I run this code it fails as it connects the `data1` to `data2`

```sv
plugin -i slang
read_slang <<EOT
module WiredNet (
    input wire data1,
    input wire data2,
    output wire out_wired
);
    wor w_or;
    assign w_or = data1;
    assign w_or = data2;
    assign out_wired = w_or;
    
endmodule
EOT

prep -top WiredNet
check -assert
```

If I ask to `write_verilog` I get : 
```sv
/* Generated by Yosys 0.54 (git sha1 db72ec3bd, g++ 12.2.0-14+deb12u1 -fPIC -O3) */

module WiredNet(data1, data2, out_wired);
  input data1;
  wire data1;
  input data2;
  wire data2;
  output out_wired;
  wire out_wired;
  wire w_or;
  assign data2 = data1;
  assign out_wired = data1;
  assign w_or = data1;
endmodule
```

povik, 2025-07-18:
Thanks for the report. Wor should be unsupported but if they are silently mishandled that's a bug

povik, 2025-08-29:
I have pushed [c962a6e](https://github.com/povik/yosys-slang/commit/c962a6e68f8cff48bf5d4d84c371eee883567803) which should catch this.

```
Top level design units:
    WiredNet

<inlined>:6:9: error: net type 'wor' unsupported
    wor w_or;
        ^

Build failed: 1 error, 0 warnings
```

I think we can change this into a feature request to have `wor` and friends supported.