povik/sv-elab #227: Possible regression in interface handling

phsauter, 2025-09-15:
When trying to synthesize Cheshire using Yosys and yosys-slang there seems to be a regression in how interfaces are handled.  
Specifically this must have happened between [IIC-OSIC 2025.03](https://github.com/iic-jku/IIC-OSIC-TOOLS/releases/tag/2025.03) (yosys-slang commit `8f2239b2b12dac43f8f47f56deef1095d5262fa2`) and [IIC-OSIC 2025.07](https://github.com/iic-jku/IIC-OSIC-TOOLS/releases/tag/2025.07) (yosys-slang commit `f106ae3a68155ee470d51b168c72282a1453171b`).

### Files
You( can download a reproducable (very much not minimal) here: http://iis-people.ee.ethz.ch/~phsauter/github_sharable/cheshire.tar.xz

This is equivalent to this PR with all the mentioned fixes already applied and an additional `--compat vcs`:
https://github.com/pulp-platform/cheshire/pull/181

### Before
When synthesizing using the provided script (navigate to `target/yosys` and run `./synthesize`) with 2025.03 and older it parses and elaborates the design just fine.

### After
When synthesizing using the provided script (navigate to `target/yosys` and run `./synthesize`) with 2025.07 or upstream:
```
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.54 (git sha1 db72ec3bd, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3)

-- Executing script file `yosys.ys' --

1. Executing Liberty frontend: synthetic.lib
Imported 13 cell types from liberty file.

2. Executing SLANG frontend.
Top level design units:
    cheshire_synth_wrapper

../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:88:34: error: hierarchical reference to unknown module instance 'reg_bus' is not allowed
  assign clic_req.addr  = reg_bus.addr;
                          ~~~~~~~^~~~~
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:62:5: note: declared here
  ) reg_bus (clk_i);
    ^
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:89:34: error: hierarchical reference to unknown module instance 'reg_bus' is not allowed
  assign clic_req.write = reg_bus.write;
                          ~~~~~~~^~~~~~
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:62:5: note: declared here
  ) reg_bus (clk_i);
    ^
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:90:34: error: hierarchical reference to unknown module instance 'reg_bus' is not allowed
  assign clic_req.wdata = reg_bus.wdata;
                          ~~~~~~~^~~~~~
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:62:5: note: declared here
  ) reg_bus (clk_i);
    ^
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:91:34: error: hierarchical reference to unknown module instance 'reg_bus' is not allowed
  assign clic_req.wstrb = reg_bus.wstrb;
                          ~~~~~~~^~~~~~
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:62:5: note: declared here
  ) reg_bus (clk_i);
    ^
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:92:34: error: hierarchical reference to unknown module instance 'reg_bus' is not allowed
  assign clic_req.valid = reg_bus.valid;
                          ~~~~~~~^~~~~~
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:62:5: note: declared here
  ) reg_bus (clk_i);
    ^
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:94:17: error: hierarchical reference to unknown module instance 'reg_bus' is not allowed
  assign reg_bus.rdata = clic_rsp.rdata;
         ~~~~~~~^~~~~~
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:62:5: note: declared here
  ) reg_bus (clk_i);
    ^
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:95:17: error: hierarchical reference to unknown module instance 'reg_bus' is not allowed
  assign reg_bus.error = clic_rsp.error;
         ~~~~~~~^~~~~~
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:62:5: note: declared here
  ) reg_bus (clk_i);
    ^
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:96:17: error: hierarchical reference to unknown module instance 'reg_bus' is not allowed
  assign reg_bus.ready = clic_rsp.ready;
         ~~~~~~~^~~~~~
../../.bender/git/checkouts/clic-360fff7ce2d41116/src/clic_apb.sv:62:5: note: declared here
  ) reg_bus (clk_i);
    ^
../../.bender/git/checkouts/cva6-20c9d7cbe0dd6995/core/cache_subsystem/cva6_icache.sv:424:58: warning: cannot select range of 64 elements from 'logic[3:0]' [-Wrange-width-oob]
    assign cl_user[i] = cl_ruser[i][{cl_offset_q, 3'b0}+:FETCH_USER_WIDTH];
                                                         ^~~~~~~~~~~~~~~~
../../.bender/git/checkouts/cva6-20c9d7cbe0dd6995/core/cache_subsystem/cva6_icache.sv:442:58: warning: cannot select range of 64 elements from 'logic[3:0]' [-Wrange-width-oob]
      dreq_o.user = mem_rtrn_i.user[{cl_offset_q, 3'b0}+:FETCH_USER_WIDTH];
                                                         ^~~~~~~~~~~~~~~~
  in instance: cheshire_synth_wrapper.i_cheshire_soc
../../hw/cheshire_soc.sv:891:5: warning: port 'next_dm_addr_i' has no connection [-Wunconnected-port]
  ) i_dbg_dm_top (
    ^
  in instance: cheshire_synth_wrapper.i_cheshire_soc
../../hw/cheshire_soc.sv:891:5: warning: port 'ndmreset_ack_i' has no connection [-Wunconnected-port]
  ) i_dbg_dm_top (
    ^

Build failed: 8 errors, 4 warnings
ERROR: Compilation failed

```

The offending code can also be seen here: https://github.com/pulp-platform/clic/blob/master/src/clic_apb.sv#L88

povik, 2025-10-17:
In the spec there's this sentence after the interface syntax is introduced:

> The interface construct provides a new hierarchical structure. It can contain smaller interfaces and can be
passed through ports.

This makes me think direct access is legal but it's not unambiguous. I'll investigate more to see what broke in your design. The example below seems to work with today's yosys-slang:

```
interface bus(input clk);
	logic a, b;
	modport primary(input a, output b, input clk);
	modport secondary(input b, output a, input clk);
endinterface

module submod(bus.primary intf);
	assign intf.b = !intf.a;
endmodule

module top(input logic w1, output logic w2,
		   input logic clk);
	bus intf(clk);
	assign intf.a = w1;
	assign w2 = intf.b;
	submod sm(intf);
endmodule
```

povik, 2025-11-12:
I've cracked it. The above example works, but the following one when specifying `--top top2` doesn't:

```
interface bus(input clk);
	logic a, b;
	modport primary(input a, output b, input clk);
	modport secondary(input b, output a, input clk);
endinterface

module submod(bus.primary intf);
	assign intf.b = !intf.a;
endmodule

module top(input logic w1, output logic w2,
		   input logic clk);
	bus intf(clk);
	assign intf.a = w1;
	assign w2 = intf.b;
	submod sm(intf);
endmodule

module top2();
endmodule
```

povik, 2025-11-12:
Filed MikePopoloski/slang#1571 upstream