povik/sv-elab #229: x values from partially initialised unpacked structs

mndstrmr, 2025-09-19:
The following produces the constant `1'x` as the value of 'o', when it should be `1'1`. This is a minimized test case from what was originally ~60,000LOC. As far as I know it exhibits no undefined behavior, since the uninitialized `a` is never accessed.  

```systemverilog
module top(
    output logic o
);
    typedef struct {
        logic tag;
        logic a;
    } t_res;

    function automatic t_res value();
        t_res r;
        r.tag = 1'b1;
        return r;
    endfunction

    always_comb begin
        t_res x;
        x = value();
        o = x.tag == 1'b1;
    end
endmodule
```

Any of the following changes solve the issue, instead producing `1`:
1. Initializing `r.a` in `value()`, or removing the `a` field from `t_res`.
2. Making `t_res` packed.
3. Replacing `x = value()` with `x.tag = 1'b1` in the `always_comb` block.
4. Merging `tag` and `a` into a single two bit field, but still only assigning/testing the first bit.

povik, 2025-09-19:
Being a correctness bug this has high priority

closed 2025-09-22 by povik

mndstrmr, 2025-09-22:
That's great, thanks!